MarlinFirmware/Marlin #25348: [BUG] G29 P1 stops at point 57 sometimes 37/100

JAbelP, 2023-02-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I saw here [that someone had a similar issue](https://github.com/MarlinFirmware/Marlin/issues/15933). I tried following the solution mentioned at the bottom by making sure that my probing_margin (#define PROBING_MARGIN 45) was less then my x and y probe offset(#define NOZZLE_TO_PROBE_OFFSET { -40, -10, -1.85 }). I didn't change my MESH_INSET (it is currently at 1). I have an ender 3 Pro with SKR Mini E3 V3.0 and a BLTouch.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10660197/Marlin.zip)

### Bug Timeline

It was mentioend before but the fixes are not working this time around.

### Expected behavior

to Finish probing

### Actual behavior

it stops at 37 or 57

### Steps to Reproduce

1. set the probe z offset
2. tram the bed corner
3. run  M190 S60 
4. run  G28 P1
5. Error around point 57

### Version of Marlin Firmware

2.1

### Printer model

Ender 3 pro

### Electronics

 SKR Mini E3 V3.0

### Add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

JAbelP, 2023-02-06:
I tried following the steps here but once I do a G29 nothing happens SENDING:G29
echo:G29
echo:M105
echo:M105

JAbelP, 2023-02-06:
I managed to run UBL from the menu and it got through the 100 points, but the output mentions that the probing failed
BLTouch Command :90
bltouch.stow_proc() end
>>> do_blocking_move_to  X222.22 Y114.56 Z11.85
>  X222.22 Y114.56 Z11.85
<<< do_blocking_move_to  X222.22 Y114.56 Z11.85
Error:Probing Failed
[ERROR] Error:Probing Failed

measured_z: 2147483647.21474836472147483647
<<< Probe::probe_at_point  X222.22 Y114.56 Z11.85
  current_position= X222.22 Y114.56 Z11.85 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X222.22 Y114.56 Z11.85
>  X197.00 Y137.00 Z11.85
echo:busy: processing
<<< do_blocking_move_to  X197.00 Y137.00 Z11.85

JAbelP, 2023-02-08:
I could still use help with this...

Bob-the-Kuhn, 2023-02-08:
The Probing Failed error is declared when the nozzle goes below Z_PROBE_LOW_POINT without getting a trigger.

Try adjusting Z_PROBE_LOW_POINT.

If that doesn't fix it the replace your BLTouch.

Tehvan, 2023-02-09:
It always stops at 37 for me. And I specifically set my Z_PROBE_LOW_POINT to -2 because I hadn't trammed my bed yet. Bed now trammed but same problem. I thought it was normal. Same thing happened to me with bltouch and tronxy black sensor.

Tehvan, 2023-02-10:
If you are getting an error it should be a
A) You haven't trammed yet
B) You have not set Z_PROBE_LOW_POINT big enough for how warped your bed is
C) Your probe is not consistent. Especially if it is an inductive probe. It may be your probe mount has drooped over time. Or there was a temperature swing. Or you are trying to probe inductively while you have tons of eddy currents in the bed because you have the heater turned on because people recommended to do that.

If you are not getting an error, don't worry about it. The 100 assumes all points probed. The autoprobing is limited by the probe margin you set.

jamespearson04, 2023-03-03:
@jAbelP It's because of how you have your ```PROBING_MARGIN``` and ```MESH_INSET``` set.

Firstly, it's normal in UBL not to probe all points, some are filled in after probing using ```G29 P2``` or ```G29 P3```. The mesh inset specifies the area that your leveling mesh covers, and the probing margin specifies the area where your probe is allowed to go. If your probe margin is bigger, as yours is, it means it's not allowed to probe any of the outer points of the mesh, so you want ```PROBING_MARGIN``` smaller than, or the same as ```MESH_INSET```.

Other points won't be reachable because of the probe-nozzle offset, but you can't do anything about that, which is why we use ```G29 P2``` or ```G29 P3```. 

You really need to read the startup guide to use UBL, it's not as straightforward as the other types: https://marlinfw.org/docs/features/unified_bed_leveling.html#ubl-startup-guide

jamespearson04, 2023-03-03:
@JAbelP Did the maths, on a 10x10 grid, the outer two loops of the mesh, which you wouldn't be able to reach with your mesh and probing margin settings, account for 64 points, which fits with you only probing 36 of them (the 37th doesn't get done).

If you reduce the probing margin so that you only miss the outer loop, that accounts for 36 points, and the nozzle to probe offset will mean you can't reach two columns of points along one side, which since you're already missing the outer 36, only costs you 8 more points, totalling 44 points. Which fits with you probing 56 of them (again 57th doesn't get done).

Basically it's working as it should.

closed 2023-05-14 by github-actions[bot]