MarlinFirmware/Marlin #25350: [BUG] Delta homing fails

Bob-the-Kuhn, 2023-02-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Ever since PR #24951, delta printers may fail to home.

On my system, the fail happens when the head needs to move a long distance.  The motors start moving and then all motors start buzzing, all motion stops and a homing error is declared.

The only work around I've found is to reduce the feedrates in HOMING_FEEDRATE_MM_M.

The git diff tool was used to narrow the failure point to PR #24951
```
89334caa526f2d300eee834d34d06d8f837a57d5 is the first bad commit
commit 89334caa526f2d300eee834d34d06d8f837a57d5
Author: tombrazier <68918209+tombrazier@users.noreply.github.com>
Date:   Mon Nov 28 03:38:15 2022 +0000

    ⚡️ Input Shaping improvements (#24951)
```

I looked through the PR #24951 changes and couldn't find anything obviously associated with delta printers.

### Bug Timeline

PR #24951

### Expected behavior

Home properly using the HOMING_FEEDRATE_MM_M settings that worked prior to PR #24951.

### Actual behavior

On my system, the fail happens when the head needs to move a long distance.  The motors start moving and then all motors start buzzing, all motion stops and a homing error is declared.

### Steps to Reproduce

1. Create a delta printer configuration.
2. Move the head by 100mm.
3. Issue G28

### Version of Marlin Firmware

bugfix-2.1.x from this morning

### Printer model

Tevo Little Monster

### Electronics

MKS_SBASE

### Add-ons

Duet Smart Effector

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/10669526/Configs.zip)

Bob-the-Kuhn, 2023-02-08:
I replaced stepper.h and stepper.cpp with pre-shaping versions.  Now it works as expected.

Bob-the-Kuhn, 2023-02-08:
In stepper.h, reverting ISR_STEPPER_CYCLES back from 100UL to 16UL makes the latest bugfix-2.1.x work as expected.

cbagwell, 2023-02-08:
#25335 should help; if not totally fix; for this case when input shaping is disabled at compile time.

Bob-the-Kuhn, 2023-02-08:
That also fixes it.

thisiskeithb, 2023-02-08:
Closing since the linked PR fixes this issue:

- https://github.com/MarlinFirmware/Marlin/pull/25335

closed 2023-02-08 by thisiskeithb