MarlinFirmware/Marlin #25351: Filament runout sensor turning off

jmger, 2023-02-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I installed the BTT Filament Runout Sensor. I just installed stable version 2.1.2. When I go to print. I have Filament RUNOUT turned on. It goes to unload the filament then I reload filament. Great so far. I then have the option on my printer a Creality Ender 3 Max with BTT Octopus Pro F429 main board. On the display it shows PURGE or FILAMENT ON as my only two options. There is no Continue Option. If I click on Filament Option I get another screen that has a Continue Option. But I have discovered that by pressing the Filament Option that disables, set the sensor to off. If I then go back in and manually turn the sensor on again then the sensor goes into a M600 and wants to unload and purge more filament. Ending up in a loop. This was happening in Version 2.0.x also. I wondered why the BTT filament sensor was not working at all after I had a filament runout.

### Bug Timeline

Sort of both a new issue reporting but this happens in the older 2.0.x marlin firmware

### Expected behavior

I expect to see a Continue button option in the on screen menu. 

### Actual behavior

As described above the Filament Runout sensor turns OFF

### Steps to Reproduce

Start a print and let the Filament runout sensor load the filament then the error or issue should show up in the menu screen that is displayed.

### Version of Marlin Firmware

I am away from my printer at this time

### Printer model

Ender 3 Max with BTT Octopus Pro 429 main board

### Add-ons

BTT Octopus Pro 429 BTT filament runout sensor

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Let me know and I can upload my con fig later when I get home if needed.

jmger, 2023-02-27:
I just installed the Trianglelabs Filament Runout Sensor. To replace my BTT Smart Sensor. I know that the wiring to my BTT Octopus Pro 429 board and the Trianglelabs sensor is installed correctly. TEST: of switch. I get a power light GREEN. I get no RED light when filament is inserted. And I get a blinking red light when filament is removed. So that is all working correctly. I haven't check the state level, yet. I still get... 1) a filament runout at the beginning of every print. And I still get the screen as described in my above post. I tried disabling the command for motion/movement of the filament. Since this sensor does not have this option. And I thought that might be the issue. No change from what I described above. Is there something else that I might have enabled that I need to disable, possibly? I have the FIL_RUNOUT_STATE set to LOW. One question: what triggers a filament Runout at the beginning of my print? I looked in my start Gcode in CUra but I don't see any commands.

PoisonLavaLucas, 2023-04-09:
I have this same issue with the firmware off the creality website on my ender 3 pro. did you ever find a solution?

closed 2023-04-09 by thisiskeithb