MarlinFirmware/Marlin #25352: [BUG] SKR 3, TMC 2208/2209, steppers buzz but do not move

mmotley999, 2023-02-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Just installed an SKR 3 (normal one, non-EZ) with the STM32H734 chip with TMC 2208s on X, Y, and Z, and 2209s on the extruder.  This was an upgrade from an SKR 1.3 and I'm using the same stepper drivers (BTT modules) that I was using on the SKR 1.3 (I moved them over to the new board).

I'm using UART and have verified the jumpers, and there is communication to the drivers (see below).

The steppers on my Y, Z, and extruder axis are not functioning.  When I go to lift Z, the steppers just buzz with no movement, Y makes an awful buzzing noise but does move the bed in a stuttery/slow manner.  Extruder (running TMC 2209) just buzzes too, and I can see the gear going back and forth rapidly (E3D Aero).   Running G28 will time out on the Y and halt the printer.

I've verified everything is physically moving smoothly, and no changes were made to the printer other than the board replacement/brain surgery.  Printer has been working perfectly for years on the SKR 1.3 (LPC1768).

TMC drivers are communicating.  I've tried changing the current up and down via M906 to no avail.   Also, disabling StealthChop does not help (just buzzes louder).

```
Send: M122

				X	Y	Z	E
 Address					0
 Enabled		false	false	false	false
 Set current	800	800	900	950
 RMS current	795	769	898	898
 MAX current	1121	1084	1266	1266
 Run current	25/31	5/31	6/31	6/31
 Hold current	20/31	4/31	4/31	4/31
 CS actual	20/31	4/31	4/31	4/31
 PWM scale
 vsense		1=.18	1=.18	1=.18	1=.18
 stealthChop	true	true	true	false
 msteps		16	16	16	16
 interp		true	true	true	true
 tstep		max	max	max	max
 PWM thresh.
 [mm/s]
 OT prewarn	false	false	false	false
 triggered
  OTP		false	false	false	false
 pwm scale sum	23	5	5	5
 pwm scale auto	0	0	0	0
 pwm offset auto	36	36	36	36
 pwm grad auto	14	14	14	14
 off time	3	3	3	3
 blank time	24	24	24	24
 hysteresis
  -end		-1	-1	-1	-1
  -start		1	1	1	1
 Stallguard thrs				0
 uStep count	40	40	40	40
 DRVSTATUS	X	Y	Z	E
 sg_result				0
 stst
 olb					*
 ola					*
 s2gb
 s2ga
 otpw
 ot
 157C
 150C
 143C
 120C
 s2vsa
 s2vsb
 Driver registers:
 		X	0xC0:14:00:00
 		Y	0xC0:04:00:00
 		Z	0xC0:04:00:00
 		E	0x80:04:00:C0
 
 
 Testing X connection... OK
 Testing Y connection... OK
 Testing Z connection... OK
 Testing E connection... OK
 ok P63 B63
```

### Bug Timeline

New with SKR 3

### Expected behavior

For the steppers to move.

### Actual behavior

Steppers, other than X, just buzz and do nothing.

### Steps to Reproduce

Simply attempt to move an axis.

### Version of Marlin Firmware

2.1.2 and also 2.1.x-bugfix as of 2/4/2023.

### Printer model

Creality CR-10S

### Electronics

Upgrade Board (SKR 3)

### Add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/10670160/config.zip)

ellensp, 2023-02-07:
STM32H734 chip?  dyslexic or a new chip? 

Its meant to be a STM32H743V

ellensp, 2023-02-07:
Please enable EEPROM_INIT_NOW to force the eeprom to be reset and try again

Bob-the-Kuhn, 2023-02-07:
Sounds like either the acceleration or the feedrates are too high.  Try dropping them to 1/4 of the current values.  

Could also be caused by the motor current being too low.

mmotley999, 2023-02-07:
Yes sorry, my dyslexia acted up (and my old eyes didn't catch it).  It is indeed the STM32H743V chip.  I only mentioned it because BTT is supposedly changing to the H723 chip per their website.

Just set EEPROM_INIT_NOW, recompiled, and reflashed.  Saw a "Settings Saved" message on the display upon boot.  Still seeing skipped steps on Z, and E doesn't extrude at all (just jitters back and forth). 

I also tried swapping one of the problem stepper drivers with a spare I had, just to rule that out.  No change.

mmotley999, 2023-02-07:
I appreciate everybody's input on this.  I'm happy to test/try anything at this point!

@Bob-the-Kuhn I am using the exact same settings for acceleration/feedrates as I was using on my SKR 1.3 running Marlin 2.0.9.3, which was working fine for the past several months (and last several years on various versions of Marlin 2.0).  I'll give it a whirl and report back, but...

This is why I opened this as a potential bug, since these issues appear to be specific to the SKR 3 board.  As I mentioned, I carried over all of the core settings that were working perfectly for months on Marlin 2.0.9.3/SKR 1.3 with no skipping/layer shifts/extrusion issues at all.

Also worthy of note:  I went ahead and upgraded my older/secondary printer (Alunar/Infitary M508, Prusa I3 clone) to Marlin 2.1.2, which still has an SKR 1.3 with the same TMC driver setup (TMC 2208s on X/Y/Z and 2209 on E) and there are NO stepper/extruder issues at all on that machine after the upgrade.  So, it doesn't appear to be a Marlin 2.1.2 issue so much as an issue with the SKR 3 / STM32H743V processor.

mmotley999, 2023-02-07:
One other thing:  Would this setting be worthy to try?  I don't know much about it....

```
/**
   * Beta feature!
   * Create a 50/50 square wave step pulse optimal for stepper drivers.
   */
  //#define SQUARE_WAVE_STEPPING
```

Bob-the-Kuhn, 2023-02-08:
Try the change in PR #25335.  It fixed my stepper buzzing problems when running normal feedrates.

mmotley999, 2023-02-08:
Thanks to ALL of you for your help so far!

@Bob-the-Kuhn, tested your fix with Input Shaping (IS) enabled and then disabled in Configuration_adv.h.

I'm not a github expert, so I looked at the change.  Looks like only one line changed in 'stepper.h'?  I remarked the old one and pasted in the new one, see below.  Did a Clean and a Build.   Here's the change I made to 'stepper.h':

```
// Model input shaping as an extra loop call
// #define ISR_SHAPING_LOOP_CYCLES(R) ((TERN0(HAS_SHAPING, ISR_LOOP_BASE_CYCLES) + TERN0(INPUT_SHAPING_X, ISR_X_STEPPER_CYCLES) + TERN0(INPUT_SHAPING_Y, ISR_Y_STEPPER_CYCLES)) * (R) + (MIN_ISR_LOOP_CYCLES) * (R - 1))
#define ISR_SHAPING_LOOP_CYCLES(R) TERN0(HAS_SHAPING, ((ISR_LOOP_BASE_CYCLES) + TERN0(INPUT_SHAPING_X, ISR_X_STEPPER_CYCLES) + TERN0(INPUT_SHAPING_Y, ISR_Y_STEPPER_CYCLES)) * (R))
```

X has always worked well, Y did skip a few steps initially (both with IS enabled and disabled).   However Z is still skipping steps, mostly when going UP versus coming DOWN.   X has low mass, Y has more mass (moving bed), and Z is overcoming gravity.... so it seems like the steppers are lacking torque for some reason?  Stepper timing issue with the faster processor... just thinking out loud.  I'm certainly no expert on the internals of Marlin!

Also of note, I received a replacement SKR 3 board, just in case I happen to have received a bad board somehow.  I've already swapped out the stepper drivers to rule that out.   I'll swap the board this evening, do some testing, and report back.

Bob-the-Kuhn, 2023-02-08:
Have you tried to increase the Y and Z motor currents?

mmotley999, 2023-02-08:
Yep. Went as high as 1200ma (M906 Z1200), same with extruder, no luck.   I also tried reducing current down (to 500) to see if perhaps it was an overdriving issue, that was worse.  Also tried M122 I (to initialize).

I even tried removing StealthChop (M569 S0 Z) to see if that would help.  That really made them buzz!

mmotley999, 2023-02-09:
Replaced board.   No change.  Checked power supply while I was there, bang on 12V.

Bumped Z back up to 1200ma and it seems to be working better, although I do seem to hear a little skip when I first move up.  Odd that I'd have to bump it up so high, as it has worked flawlessly at 900ma for literally years, including right before I swapped the SKR 1.3 with the SKR v3.

Bumped extruder to 1200ma as well and heated it up.  It buzzes until I "help it" by flicking the the main gear with my finger (it's a geared extruder, E3D Aero).  Without that help, it just buzzes/toggles back and forth.  Still something going on there.  It's a genuine E3D slimline stepper.

Bob-the-Kuhn, 2023-02-09:
Did you ever try reducing the feedate and/or the acceleration?

mmotley999, 2023-02-09:
I hadn't before, but I just did. No change unfortunately.  I focused on Z axis testing.

Did I pick the right gcode for feedrate? Was debating M220 (percentage), which I can try as well if that would help.

Acceleration Change:

Original Setting (M503):  
M204 P600.00 R1000.00 T800.00
Test Setting:  
M204 P200.00 T200.00

Feedrate:

Original (from M503):  
M203 X2500.00 Y2500.00 Z100.00 E25.00
Test Setting:  
M203 Z25.00

Seems like the Z motors buzzed louder after these settings when lifting the gantry.  

(FYI I kept the TMC current setting at my normal setting of 900ma so I was only changing one thing at a time)

mmotley999, 2023-02-11:
UPDATE:  I just downgraded, configuring and installing the latest **Marlin 2.0.x-bugfix** release, and both the Z steppers AND the extruder steppers are working perfectly.  No buzzing, no skipping.  Using the same TMC current settings I've used previously for all axis.

So, this seems to only be an issue when running Marlin 2.1.x.

FYI, to get 2.0.x working well, I had to tweak a few things in ini/stm32h7.ini as reports on the BTT github issue log are that TMC UART doesn't work well without an updated arduinoststm32 (compiled cleanly and can confirm M122 is showing all ok):

```
#
# BigTreeTech SKR V3.0 / V3.0 EZ (STM32H743VIT6 ARM Cortex-M7)
#
[env:STM32H743Vx_btt]
extends                     = stm32_variant
platform                    = ststm32@~14.1.0
# REMARK platform_packages           = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip
platform_packages           = framework-arduinoststm32@~4.20200.221104
```

Also had to change the "f_CPU" in the "marlin_STM32H743Vx.json" file, as it still had 400Mhz defined (bumped it 480Mhz to match 2.1.x).

I'm willing to go back to 2.1.2 if any of the developers need debugging information and can provide guidance... I'm happy to contribute how I can.

thisiskeithb, 2023-02-11:
> FYI, to get 2.0.x working well, I had to tweak a few things in ini/stm32h7.ini as reports on the BTT github issue log

This has already been corrected in `bugfix-2.1.x`:

- https://github.com/MarlinFirmware/Marlin/pull/24711

https://github.com/MarlinFirmware/Marlin/blob/ef3f484d3ca620272f849b913851541b54b51c09/ini/stm32h7.ini#L42-L48

---

> Also had to change the "f_CPU" in the "marlin_STM32H743Vx.json" file, as it still had 400Mhz defined (bumped it 480Mhz to match 2.1.x).

This has already been corrected in `bugfix-2.1.x`:

- https://github.com/MarlinFirmware/Marlin/pull/24721

https://github.com/MarlinFirmware/Marlin/blob/ef3f484d3ca620272f849b913851541b54b51c09/buildroot/share/PlatformIO/boards/marlin_STM32H743Vx.json#L6

---

I just compiled an SKR 3 build and here's the top of the build log with package versions:
<details><summary>SKR 3 Build Details</summary>
<p>

```prolog
Processing STM32H743Vx_btt (platform: ststm32@~15.4.1; board: marlin_STM32H743Vx; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32H743Vx.html
PLATFORM: ST STM32 (15.4.1) > STM32H743VI (1024k RAM. 2048k Flash)
HARDWARE: STM32H743VIT6 480MHz, 1MB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 @ 4.20200.221104 (2.2.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|   |-- SPI @ 1.0
|   |-- SoftwareSerial @ 1.0
|-- LiquidCrystal @ 1.5.1
|   |-- Wire @ 1.0
|-- SlowSoftI2CMaster @ 1.0.0
|-- SlowSoftWire @ 0.0.0+20220807061740
|   |-- SlowSoftI2CMaster @ 1.0.0
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Wire @ 1.0
|-- Servo @ 1.1.2
Building in release mode
...
```

</p>
</details>

Using a CI script in GitHub Actions, I also see it is also pulling down `framework-arduinoststm32 @ 4.20200.221104`, so I suspect that an older arduinoststm32 platform package is cached on your system. PIO should pull down 4.20200.221104 if you're using the latest `bugfix-2.1.x` from this repo.

Some things to try with the latest `bugfix-2.1.x`:

1. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

2. Re-install PlatformIO.

3. Build your config with the latest `bugfix-2.1.x` without changing `stm32h7.ini`. Copy the top of the build log with package versions like I have above and paste that in your next reply.

I'll try to scrounge up some 2208s to recreate your exact hardware config on an SKR 3 for testing, but when I run with TMC2209s, I don't have any motion/buzzing issues.

mmotley999, 2023-02-11:
@thisiskeithb the changes I made were to the bugfix-2.0.x branch, not to bugfix-2.1.x.  I made NO changes when compiling 2.1.2 or bugfix-2.1.x as those files were current per your PR references.... those files are not current in the bugfix-2.0.x branch:

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/ini/stm32h7.ini
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/PlatformIO/boards/marlin_STM32H743Vx.json

Since I opened the issue, I went ahead and replaced my 2208s with BTT 2209 v1.3 modules (might as well).  Stepper issue persisted.

Per your request, downloaded the latest bugfix-2.1.x, deleted the '.platformio' directory, restarted VSC (I'm on Windows, BTW) which reinstalled PlatformIO.  Compiled successfully, loaded it on the printer, and still Z and E are buzzing.

Here's the output.  Also including the config files I used at the bottom.

```
Processing STM32H743Vx_btt (platform: ststm32@~15.4.1; board: marlin_STM32H743Vx; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32H743Vx.html
PLATFORM: ST STM32 (15.4.1) > STM32H743VI (1024k RAM. 2048k Flash)
HARDWARE: STM32H743VIT6 480MHz, 1MB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.20200.221104 (2.2.0)
 - framework-cmsis @ 2.50700.210515 (5.7.0)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|   |-- SoftwareSerial @ 1.0
|   |-- SPI @ 1.0
|-- U8glib-HAL @ 0.5.2
|   |-- Wire @ 1.0
|   |-- SPI @ 1.0
|-- SlowSoftI2CMaster @ 1.0.0
|-- SlowSoftWire @ 0.0.0+20230211115254
|   |-- SlowSoftI2CMaster @ 1.0.0
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Wire @ 1.0
|-- Servo @ 1.1.2
Building in release mode

```
[Configuration-bugfix-21x.zip](https://github.com/MarlinFirmware/Marlin/files/10714601/Configuration-bugfix-21x.zip)

thisiskeithb, 2023-02-11:
Your configs are not current `bugfix-2.1.x` because they're missing a lot of changes.

Why did you change RSENSE values to 0.011 & 0.062? They should be 0.11 with TMC2208/9s:

[image]

Here are the updated configs for current `bugfix-2.1.x` (includes correct RSENSE values): [25352-configs.zip](https://github.com/MarlinFirmware/Marlin/files/10714831/25352-configs.zip)

Here's a compiled bin from the configs above that do not have any UART or stuttering issues on any axis (using Creality motors): [25352-bin.zip](https://github.com/MarlinFirmware/Marlin/files/10714825/25352-bin.zip)

mmotley999, 2023-02-12:
@thisiskeithb may I have some salt with the crow I'll be eating?  :-)

The RSENSE was the issue.  I changed from "0.011" to  "0.11", recompiled, and all steppers are working normally.  The .011 was my mistake (the template I downloaded had the "0.062" listed, so just changed "62" to "11").  I can't believe I didn't catch that, as I did it correctly on bugfix-2.0.x version and have compiled Marlin many, many times over the years.

Thank you all for your help and input.  My apologies for this being my mistake.

closed 2023-02-12 by mmotley999