MarlinFirmware/Marlin #25353: Erratic Behavior and Loss of Steppers

jmb-555, 2023-02-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The latest stable marlin was just flashed to a stock Ender 5 Pro with the stock ender 5 config files and setup. A gcode extrusion calibration file was tested and the machine was not making x or y movements even though its clearly in the gcode files. The machine basically sat there and aggressively gushed out filament at an extrusion rate of probably 100X normal flow rate. The extruder was skipping worse than anything I've seen in my 8-9 years of 3d printing. The file was printed again and the same exact thing happened. Manual gcode was sent in from the Arduino IDE over serial. The machine replied back "Ok" to both x and y gcode commands but there was no movement. The machine was restarted and everything worked as expected. The previously mentioned gcode file printed as expected. I wish i had a log file or something to attach because it seems like it would help. The config files are attached as well as the gcode that gave the issues. Seem unlikely that the stock Ender 5 profile would be giving me trouble like that but it felt necessary to report.
[BugFiles.zip](https://github.com/MarlinFirmware/Marlin/files/10670787/BugFiles.zip)

### Bug Timeline

Issue on flash. Started first print

### Expected behavior

Expected behavior above

### Steps to Reproduce

I have not reproduced this the reboot solved the issue.

### Version of Marlin Firmware

2.1.x

### Printer model

Ender 5 Pro

### Electronics

Stock

### Add-ons

No add ons

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I didn't know about M111 S247 logging. Ill will make sure to get a good log submitted next time.

ellensp, 2023-02-07:
Normal cause is failing to reset eeprom data to defaults after a firmware upgrade.

M502 then M500

or enable 
  #define EEPROM_INIT_NOW   // Init EEPROM on first boot after a new build.

To update eeprom to defaults every time you flash

jmb-555, 2023-02-07:
I was prompted with a "reset" question on boot up. Im assuming that this did the reset. When the erratic extrusion occured, i looked at the saved steps in the config tab and they were all good. I think 80 for the steppers and 90 something for the extruder.

Also, there was one other gcode file printed before the one that I initially attached. If that file worked, then how could it be an EEPROM issue? That other gcode file is attached.
[ExtrusionCalibration.zip](https://github.com/MarlinFirmware/Marlin/files/10677097/ExtrusionCalibration.zip)

Bergerac56, 2023-02-17:
Just looking at your video. Could it be that your extruder is too close from the bed and block extrusion?

jmb-555, 2023-02-17:
Dont think it's the extruder being too close. It's clearly trying to squirt more plastic than it can handle. The plastic that comes out is much wider than normal. The motor is also clearly trying to spin at a ridiculous speed when lifting the extruder lever. Also, like i said the machine also locks up in x and y positions. It replies that it gets x and y commands, but it never moves. It just sits there trying to squirt out more plastic than it can handle. It seems like all of this points to software issues not hardware.

closed 2023-04-30 by github-actions[bot]