MarlinFirmware/Marlin #25355: [BUG] ? On Copymaster 3D 300 Printer.. Bed Levelling issue V2.1.2

PicAbuser, 2023-02-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Using default configuration.h for copymaster 300 (8 bit board)..
The Bed Levelling feature places the extruder over 9 points on the bed allowing for the Z axis to be tweaked so that a matrix can be built.
Sadly, the first 3 points extend past the bed so the extruder is over fresh air by about 15mm.
The bed is moved to the back of the printer. It is NOT over extended as there are a few mm of movement before hitting the end of the track. Note this is the opposite end to the limit U'switch.
Hitting the mechanical end of travel would make a hell of a noise from the steppers.

Note that TRAMMING puts the extruder perfectly centred over the bed so I THINK the parameters that set the coordinates for the bed levelling are incorrect and, of course, despite much searching, I cannot find these using "level" as a text search entry.
I HAVE located where to change the matrix from "3" (a 3x3 matrix) to "4" to create a 4x4 matrix as the bes is rather large.

The Facebook page is not responding to the question "Where are the bed levelling starting coordinates initialised"? and an image of the overhang for reference.

A BUG is probably too harsh and the copymaster is hardly a very common printer.
I updated the software from the clunky original using Visual studio code and PlatformIO as the you tube tutorials suggest.
I program PIC's exclusively in assembler and can program in C# a little so not a novice but not a day to day user of C.

A line number and a file name would help me out and some guidance as the end stop switch is at the front of the printer not at the back like an ender 3 so the parking number is too big as the bed moves a long way from the homing point

### Bug Timeline

Unknown

### Expected behavior

I expected bed levelling when the bed is furthest away from the microswitch to be the same or similar to the correct parking positions as provided by the TRAMMING feature.

### Actual behavior

As stated, the bed moves too far away from the homing point such that the extruder is looking at fresh air, not the bed.

### Steps to Reproduce

1.. select "Bed Levelling" and select "position 1".
2.. note the bed moved away from the home point, towards the rear of the printer.
3.. Note that the bed has travelled further than it should by about 15mm.
4... Select position 2 and 3 noting the over travel remains.
5.. Note that positions 7,8 and 9 are over the bed BUT, sadly, I did not measure the extruder position so there COULD be a bed offset rather than a incorrect coordinate calculation. (my bad).
6.. Abandon bed levelling and select "TRAMMING".
7.. NOTE that all 4 positions place the bed under the extruder in what seem to me to be a perfectly centred operation, hence my assertion that bed levelling definitions are potentially incorrect and NOT shared with tramming code.

I am uncertain if this is a translation shift (The whole bed is 15mm out in the Y axis), or the first bed coordinate is incorrectly defined or calculated.
With the line number of the relevant definitions, I could probably derive the numbers myself.

### Version of Marlin Firmware

2.1.2

### Printer model

Copymaster 3D 300

### Electronics

Stock. 

### Add-ons

No additional hardware. NO changes to configuration.h

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configuration.h is exactly as provided in the copymaster folder and is a solitary file in the folder.
Configuration_adv.h is as supplied when marlin is downloaded from GitHub.

I have made NO changes at all to any of the files. They are as downloaded from you. 
All I did was COPY configuration.h from the copymaster 300 folder and replace the configuration.h file in the marlin project.
Build and upload to the printer.. Job done.. Lovely.

[files.zip](https://github.com/MarlinFirmware/Marlin/files/10679805/files.zip)

EvilGremlin, 2023-02-07:
https://manuelmclure.github.io/ConfiguringLeveling.html

PicAbuser, 2023-02-08:
I am deeply embarrassed to say I have resolved the issue myself without messing with the code.
I hang my head in shame and beg forgiveness for easting valuable time.

closed 2023-02-08 by PicAbuser