MarlinFirmware/Marlin #25356: [BUG] "MESH_EDIT" error on compile

rq3, 2023-02-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling the latest (Feb.7, 2023) bugfix with the latest configuration files throws this error, even though the printer has a 480x320 graphical LCD and touchscreen:
#error "MESH_EDIT_GFX_OVERLAY requires a Graphical LCD."

### Bug Timeline

This bug has been there for years.

### Expected behavior

I expect it to compile without error, with a mesh edit gfx overlay on the graphical display.
To compile, the MESH_EDIT_GFX_OVERLAY line in UBL must be commented out in configuration.h

### Actual behavior

Compile fails with error.

### Steps to Reproduce

1) Compile latest bugfix with latest configuration files.
2) See  #error "MESH_EDIT_GFX_OVERLAY requires a Graphical LCD."

### Version of Marlin Firmware

bugfix 2.1.3

### Printer model

Anycubic Predator

### Electronics

stock Trigorilla_Pro with STM32F103

### Add-ons

none

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10679996/Configuration.zip)

thisiskeithb, 2023-02-07:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22875

thisiskeithb, 2023-02-07:
You will need to use `TFT_CLASSIC_UI` to use `MESH_EDIT_GFX_OVERLAY`.

closed 2023-02-07 by thisiskeithb

jmz52, 2023-02-08:
@rq3, MESH_EDIT_GFX_OVERLAY is a feature of graphical 12864 interface.
It requires either graphical _LCD_ or LCD emulation on TFT screen (TFT_CLASSIC_UI)

>though the printer has a 480x320 graphical LCD and touchscreen

That is a 480x320 TFT screen, not an LCD screen.
While this might cause some confusion, the error message is technically correct.