MarlinFirmware/Marlin #25363: [FR] (feature summary) M code to set machine UUID

skl111, 2023-02-10:
### Describe the feature you want

Have and Option to enable an M command to Retrieve and Set a machine's UUID

### Additional context

This would be helpful for those who want to connect multiple printers via USB to 1 host.

Here is a link to a project that would benefit from this which uses OctoPrint.

https://github.com/paukstelis/octoprint_deploy

As for why someone would want to use a single host for multiple printers.

A mini pc or similar host could take the place of multiple Raspberry Pi's which are scarce and subject to price gouging.

ellensp, 2023-02-10:
M115 returns this.information already.

It is not meant to be settable. 

```
// Printer's unique ID, used by some programs to differentiate between machines.
// Choose your own or use a service like https://www.uuidgenerator.net/version4
//#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"

```

Allowing this to be settable defeats the purpose of identification.  Ie nothing stopping you changing it to a conflicting id or even swapping id's...

skl111, 2023-02-10:
> M115 returns this.information already.
> 
> It is not meant to be software settable. 
> 
> ```
> // Printer's unique ID, used by some programs to differentiate between machines.
> // Choose your own or use a service like https://www.uuidgenerator.net/version4
> //#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
> 
> ```
> 
> Allowing this to be settable defeats the purpose of identification. A unique, unchangeable id. 
> 

Is that value stored in eeprom and imported when the machine is flashed with a new version?

ellensp, 2023-02-10:
That data in in the flash storage, part of the firmware

skl111, 2023-02-10:
> That data is in the flash storage, part of the firmware

So is it imported from the flash storage when the firmware is upgraded?

From my understanding its not.

So maybe the purpose needs to be re-evaluated as it is not being used for the intended purpose generally. 

It's generally set to the default value of cede2a2f-41a2-4748-9b12-c55c62f367ff or whatever the source repo has it set to for users who compile their own firmware from those repos.

And if your one of the many users who download recompiled bins, then you have the same uuid as well.

It also presents a different issue if your compiling firmware and you have say 2 ender 3s and 2 sv01s, you will have to keep either 4 repos with the uuids being different in each one, or 2 copies of the ender 3 uuid file and 2 of the sv01 uuid file and swap them out to keep the uuids the same every time you compile and you have to make sure to flash the right printer. 
It gets worse the more printers you have. 

If the value was stored in some fashion so it could be imported after flashing during boot, that would eliminate having to keep track of multiple UUID's just to keep the machine UUID the same when you flash new firmware.

sui77, 2023-05-25:
The UUID implementation is pretty much useless for end users who don't compile their own firmware. Both of my machines have the very same UUID and I'd also love to have a feature to set an individual name/identifier. (btw. I use a 50 bucks from ebay dell wyse 3040, which easily handles two octoprint instances.) 

However here is my workaround for now: Use the extruder acceleration setting instead. The decimal places (or even +- 1000) virtually won't make a difference and the value is reported right after connection, so you don't even need to request it.

    M201 E10000.01   // first printer
    M500             // save settings

    M201 E10000.02   // second printer
    M500             // save settings

    M201             // get value