MarlinFirmware/Marlin #25372: [BUG] Unable to get AnyCubic LCD 1.0 to work

GrevanGER, 2023-02-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Greetings. 
The problem is as follows: 

Within configuration.h there is only one segment about AnyCubic LCDs. 

```
#define ANYCUBIC_LCD_I3MEGA
//#define ANYCUBIC_LCD_CHIRON
#if EITHER(ANYCUBIC_LCD_I3MEGA, ANYCUBIC_LCD_CHIRON)
    #define LCD_SERIAL_PORT 1
  //#define ANYCUBIC_LCD_DEBUG
  //#define ANYCUBIC_LCD_GCODE_EXT  // Add ".gcode" to menu entries for DGUS clone compatibility
#endif
```

So with my i3 Mega running the stock Trigorilla 1.0 board and the AnyCubic LCD 1.0 (blue images on white background),
I am getting the following error at this point: 

`#error "Serial Port 1 pin D18 and/or D19 conflicts with another pin on the board."`

Pin 18 and 19 are part of the limit switches on this board. But since I don't know nothing about Ports, I am kinda out of my element to fix this problem. 

[image]

### Bug Timeline

Not sure

### Expected behavior

I expected the line "#define AnyCubic_LCD_I3MEGA" to work with older hardware as well.

### Actual behavior

It did not.

### Steps to Reproduce

1. Have an old Anycubic i3 Mega around
2. Try to make Marlin work

### Version of Marlin Firmware

2.1.2

### Printer model

AnyCubic i3 Mega S

### Electronics

Trigorilla 1.0 and AnyCubic 1.0 LCD

### Add-ons

none

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Bob-the-Kuhn, 2023-02-12:
By any chance do you have the pinout of the LCD?  Without this we won't know what pins are used for the serial port so we can't give you the correct LCD_SERIAL_PORT setting.

Bob-the-Kuhn, 2023-02-12:
Try deleting the line `#define LCD_SERIAL_PORT 1` and see what happens.

I just looked at the [example bugfix-2.1.x config files](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/AnyCubic/Chiron) for Anycubic Chiron and that line isn't there.

thisiskeithb, 2023-02-12:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-02-12 by thisiskeithb

GrevanGER, 2023-02-12:
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

It is a bug when a Feature is Not working as expected, isn't? I understand that with the amount of issues raised, you can't read through every Post. 
But when the Integration of a Board and Display which have been around as Long as the Marlin project stops working, than it is a bug.

GrevanGER, 2023-02-12:
> Try deleting the line `#define LCD_SERIAL_PORT 1` and see what happens.
> 
> I just looked at the [example bugfix-2.1.x config files](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/AnyCubic/Chiron) for Anycubic Chiron and that line isn't there.

Auto builder replies with: 
`#error "The ANYCUBIC LCD requires LCD_SERIAL_PORT to be defined."`

thisiskeithb, 2023-02-12:
> It is a bug when a Feature is Not working as expected, isn't

This is a configuration issue, not a bug. 

The sanity check told you pins conflict, so you need to use a different serial port: `#error "Serial Port 1 pin D18 and/or D19 conflicts with another pin on the board."`

GrevanGER, 2023-02-12:
> > It is a bug when a Feature is Not working as expected, isn't
> 
> This is a configuration issue, not a bug.
> 
> The sanity check told you pins conflict, so you need to use a different serial port: `#error "Serial Port 1 pin D18 and/or D19 conflicts with another pin on the board."`

When hardware that was mapped fine yesterday, is Not mapped correctly Today, it is an error/Bug on the Devs Teams side, not mine as a user.

Bob-the-Kuhn, 2023-02-13:
Try setting LCD_SERIAL_PORT to xx and see if any work.  Start with 2 and keep going until the compiler says you're outside the range of acceptable serial ports.

ellensp, 2023-02-13:
Serial: 0 (RX) and 1 (TX); 
Serial 1: 19 (RX) and 18 (TX); 
Serial 2: 17 (RX) and 16 (TX); 
Serial 3: 15 (RX) and 14 (TX). 

Serial 0 is your USB cable.
Serial 1, pin 18 is used as Z2_MIN_PIN so cannot be used
Serial 2  not very likely, doesn't look like adapter pcb plugs into exp1 where these pins are.  
Serial 3. Is possible, and is the default in marlin when #define LCD_SERIAL_PORT is not set. 

Ie remove   #define LCD_SERIAL_PORT 1 from your configuration.h file.

GrevanGER, 2023-02-13:
> > It is a bug when a Feature is Not working as expected, isn't
> 
> This is a configuration issue, not a bug.
> 
> The sanity check told you pins conflict, so you need to use a different serial port: `#error "Serial Port 1 pin D18 and/or D19 conflicts with another pin on the board."`

Seeing you adding the "False alarm" tag right right ... dang... I wish I could grab and shake you some. 

The hardware was working fine with previous versions. 
Someone screwed up and now I need to fix that. That is not part of the intended user experience aka "configuration error". 
Your team put the "#define LCD_SERIAL_PORT 1" there, not me. 

And you are setting there "Nope, your fault, not ours that someone from our team made a mistake there."

GrevanGER, 2023-02-13:
> Try setting LCD_SERIAL_PORT to xx and see if any work. Start with 2 and keep going until the compiler says you're outside the range of acceptable serial ports.

For serial port 2: 
No  error when compiling, but also not working when being loaded on to the printer. 
(And yes, I reboot the machine after flashing the firmware to make sure it loads the new one.)

For serial port 3: 
`#error "Serial Port 3 pin D14 and/or D15 conflicts with another pin on the board."`

For serial port 4: 
`#error "LCD_SERIAL_PORT must be from 0 to 3."`

So the error is within the mapping of Port 2.

GrevanGER, 2023-02-13:
When going through the ramps>pins_Trigorilla_14.h there is no section about ports.
But I found this: 
`//
// AnyCubic pin mappings
//
// Define the appropriate mapping option in Configuration.h:
// - TRIGORILLA_MAPPING_CHIRON
// - TRIGORILLA_MAPPING_I3MEGA
//`

Now guess what. There is no TRIGORILLA_MAPPING in the Configuration.h
I can only assume that this referes to 
`#define ANYCUBIC_LCD_I3MEGA
//#define ANYCUBIC_LCD_CHIRON`
Which is also no help, since these option only define the ports, but not the port mapping.

But lets head on. 
Since there is not a single entry about EXP1 and EXP2 in the pins_TRIGORILLA.h, I looked in to the included pins_RAMPS.h.
And what can I say? I am completely lost. 
There are a few examples of how the pin mapping is working, but there are so many cross defines and references, that I am afraid I am at risk of frying my hardware when I screw up at this point.

ellensp, 2023-02-13:
are you even using marlin provided configs?
https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.1.x/config/examples/AnyCubic/i3%20Mega/Trigorilla%20AVR/Configuration.h#L159
"#define TRIGORILLA_MAPPING_I3MEGA"

ellensp, 2023-02-13:
Marlin bugfix 2.1.x using marlin provided configs for AnyCubic/i3 Mega/Trigorilla AVR/
Compiles without issues.