MarlinFirmware/Marlin #25380: [FR] Support the newer "TJC" screens on the Ender 3 S1

dandrzejewski, 2023-02-13:
### Is your feature request related to a problem? Please describe.

I have a new Ender 3 S1 with the new "TJC" LCD.  It does not currently work with Marlin.

### Describe the feature you want

I would like support for this newer LCD screen.

### Additional context

I'm happy to help test.

ben-wallis, 2023-02-21:
There's some discussion related to the TJC/Nextion LCDs that might be of interest to anyone looking to implement this in https://github.com/MarlinFirmware/Marlin/issues/25195#issuecomment-1374433909

ben-wallis, 2023-02-21:
@dandrzejewski You might also be interested to know that while there's no LCD support for the TJC LCD that works 100%, I have Marlin compiled for the DWIN LCD (with `#define DWIN_CREALITY_LCD` set) on my S1 with a TJC LCD and it works fine with the exception of some garbled icons and text. If all you want is access to similar functionality to that of the stock Ender3 S1 firmware (movement, z-offset, pre-heat etc) then this is a decent option until full support for TJC LCDs is added.

dandrzejewski, 2023-02-22:
By the way, while I don't have time to develop this, I'm happy to provide any further details on this hardware and test anything that needs to be tested.