MarlinFirmware/Marlin #25385: Laser M3/M4 I (inline) not firing laser while standard M3 does

dee-nine, 2023-02-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Unable to get work inline mode with my "15W" diode laser (3 wires connection: +12V, GND, PWM 5V), laser just not fires when using M3/M4 I.
In standard mode M3/M4 and in test menu laser fires.
Simply i am able to cut geometry with it but not engrave grayscale bitmaps (using Lightburn - switching M03/M05 and inline mode).
PWM connected to empty servo pin (tried D6, D45 - same result) and configured in pins_RAMPS.h.

### Version of Marlin Firmware

Marlin bugfix-2.1.x

### Printer model

Own design, coreXY

### Electronics

MKS gen L v2.1 (mega2560)

### Add-ons

tmc2209+sensorless homing, bltouch, MINI 12864 V1.0 LCD, autofan on E1

### Bed Leveling

No Bed Leveling

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/10737694/configs.zip)

ellensp, 2023-02-15:
PWM is broken on ramps, awaiting merge of https://github.com/MarlinFirmware/Marlin/pull/25343

ellensp, 2023-02-16:
This PR has been merged, please try the current bugfix 2.1.x

closed 2023-02-16 by thisiskeithb

dee-nine, 2023-02-16:
Seems to working, thanx!