MarlinFirmware/Marlin #25386: [FR] LASER / SPINDLE pwm control on teensy 4.1

GHubensG, 2023-02-15:
### Is your feature request related to a problem? Please describe.

hi,
I am using marlin 2.1.2 on an SKR 1.4 to control my CO2 laser. I want to upgrade my main board to a Teensy 4.1 to have faster movements because the SKR 1.4 can't handle how much DPI and speed i want.
My problem is the fault message i get;
`In file included from Marlin\src\HAL\TEENSY40_41\../../inc/MarlinConfig.h:50:0, from Marlin\src\HAL\TEENSY40_41\HAL.cpp:29: Marlin\src\HAL\TEENSY40_41\../../inc/../HAL/TEENSY40_41/inc/SanityCheck.h:33:4: error: #error "Features requiring Hardware PWM (FAST_PWM_FAN, SPINDLE_LASER_FREQUENCY) are not yet supported on Teensy 4.0/4.1." #error "Features requiring Hardware PWM (FAST_PWM_FAN, SPINDLE_LASER_FREQUENCY) are not yet supported on Teensy 4.0/4.1." `

can someone help me to fix this or is this not possible?

### Are you looking for hardware support?

Board: Teensy 4.1

### Describe the feature you want

Hardware PWM control

thisiskeithb, 2023-02-15:
> can someone help me to fix this or is this not possible?

There's nothing to "fix" since it's not implemented yet:

`#error "Features requiring Hardware PWM (FAST_PWM_FAN, SPINDLE_LASER_FREQUENCY) are not yet supported on Teensy 4.0/4.1."`

Someone will need to code this feature & submit a PR for it to work on Teensy.

phizev, 2024-12-28:
@GHubensG I've opened a [PR](https://github.com/MarlinFirmware/Marlin/pull/27608) which should implement this feature, I do not have a machine with a laser, nor spindle to test with.
I know it's been a while, but if you are still interested in moving forward with changing to a Teensy 4.1 having someone able to test with actual hardware would greatly help.

closed 2025-01-13 by thinkyhead