MarlinFirmware/Marlin #25393: [BUG] M23/M24 over serial works but starting print from TFT doesn't

CobaltCause, 2023-02-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If I try starting a print from the onboard screen, the machine crashes. However, if I connect to it via serial, I can issue M23/M24 to start the print and everything works fine. If I connect to serial and try to start the print from the screen, I see `Error:SD read error` repeated forever until I reboot the printer. Issue persists with brand new SD card.

### Expected behavior

I expected the print to start.

### Actual behavior

The screen froze and the machine didn't do anything after selecting the check when asked for confirmation, displaying this:

[image]

If I'm connected to serial, I see `Error:SD read error` repeatedly printed. I have to reboot the printer to get it out of this state.

### Steps to Reproduce

1. Put any gcode into a file, I was using `G28\nG0 X5 Y5 Z5` for testing, then put it on the SD card
2. Insert SD card into printer and power the printer on, in any order
3. Start print using the built-in screen

### Version of Marlin Firmware

bugfix-2.1.x 38e633023603559032f2daa80b1ee99d9bd4df72

### Printer model

SliceWorx/Kingroon KP3S Pro S1

### Electronics

Kingroon v1.3 (GD32F303VET6)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

* [`Configuration.h`](https://or.computer.surgery/charles/marlinfw/-/blob/c6682b84fba57ef3f48022cf913a343d562f6ef0/Marlin/Configuration.h)
* [`Configuration_adv.h`](https://or.computer.surgery/charles/marlinfw/-/blob/c6682b84fba57ef3f48022cf913a343d562f6ef0/Marlin/Configuration_adv.h) (untouched)

Logs after issuing `M111 S247` and starting a print from the SD card using the built-in screen:

```
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION
ok
echo:Now fresh file: test~1.gco
File opened: test~1.gco Size: 16
File selected
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
Error:SD read error
# repeats forever till reboot
```

ellensp, 2023-02-16:
GD32 cpus  are not supported in Marlin. no they are not clones of STM32, they are sufficiently different to cause all sort of issues.

"The GD32 has two truncated DMA controller and it cannot serve two channels in parallel.
for SDIO and screen need to use own DMA controller."

There is a patch related to this.
https://github.com/jmz52/Marlin/commit/1940418bbe89d07863ed05c6cddb1edf285a5a31#diff-f64f057c8653ee97ebf3be2065e2d0e14ec568be6ab1e03c98b5282fbf1e4ecf

I have no idea if it works

You should continue to use official firmware provided by you manufacture, until Marlin get official GD32 support.

See ongoing discussion. https://github.com/MarlinFirmware/Marlin/discussions/24985

closed 2023-02-16 by ellensp

CobaltCause, 2023-02-22:
For posterity, cherry-picking that patch into my tree did fix the problem. I also discovered later on that, without that patch and by starting a print over serial via the mentioned G-codes, the printer would crash pretty quickly. The patch did away with that issue as well. Been messing with the printer running Marlin since your comment and haven't had any weird issues come up again (yet?). Thanks for pointing that out!