MarlinFirmware/Marlin #25396: BTT SKR3 MCU STM32H723VGT6 Support

sacy80, 2023-02-17:
### Is your feature request related to a problem? Please describe.

Hi all

I got a BTT SKR3 which the MCU are "STM32H723VGT6", i try compile Marlin 2.1.2 with env:STM32H743Vx_btt. but when firmware flash. the board not responding . no usb com. 

Possible due to MCU env not compatiable? 

### Are you looking for hardware support?

SKR3 with MCU "STM32H723VGT6"

### Describe the feature you want

MCU support for SKR3 MCU "STM32H723VGT6"

ellensp, 2023-02-17:
See https://github.com/MarlinFirmware/Marlin/pull/25387#issuecomment-1432360988
Is already being worked on, wonce BTT release the code.

thisiskeithb, 2023-02-18:
I've pinged @bigtreetech about this, but it won't be added in the referenced PR.

BigTreeTech will need to submit a PR or post the code so I can port it upstream, but this board/MCU combination won't be supported until that happens.

CristianoGorla, 2023-02-19:
Hello,
I just successfully compiled and flashed Marlin from the Bigtreetech Github Octopus-Max-EZ(https://github.com/bigtreetech/Octopus-Max-EZ).
I'm using the STM32H723VGT6 MCU version and this is the only Marlin source that boot the board.
I think you can find the things to make compatible the official Marlin version.

thisiskeithb, 2023-02-19:
I've added SKR V3.0/V3.0 EZ support back into the referenced PR:

- https://github.com/MarlinFirmware/Marlin/pull/25387

closed 2023-02-19 by thisiskeithb

CristianoGorla, 2023-02-19:
Thanks @thisiskeithb