MarlinFirmware/Marlin #25401: [BUG] The ZMAX endstop is ignored and gear grinds when using ABL

Poikilos, 2023-02-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When a BLTouch is used for homing, the gears grind for a while against the maximum. The issue is especially prominent with a printer where the bed moves up and down, since when the grease is warm and the printer turns off, the bed will fall, so when the printer turns back on it reads Z as 0 but Z is actually max. If a ZMAX sensor is installed, Marlin refuses to use it despite #10390 being fixed.

Related issues:
- #10390
- #23759

Already tried:
- It occurs whether `NOZZLE_PARK_FEATURE` is enabled or not.
- I custom mounted a ZMAX sensor on my Replicator 2X with a BTT SKR V1.4 Turbo mainboard. It lights up, but is ignored by Marlin. Everything else works fine.
- `#define USE_ZMAX_PLUG`

### Bug Timeline

Posts around the internet indicated that fixing #10390 would fix this but it did not.

### Expected behavior

I expect Marlin to stop increasing Z when the ZMAX sensor is triggered, even if the probe is used for homing and `#define Z_HOME_DIR -1`.

### Actual behavior

When the printer is first turned on, Marlin reads Z as 0 though it is higher, so running ABL or filament load causes the Z to keep increasing as the gears (worm gear and follower) grind even after the ZMAX endstop is triggered.

### Steps to Reproduce

1. Install a Probe.
2. Install a ZMax sensor.
3. Turn the printer off.
4. Move the nozzle near ZMAX.
5. Turn the printer on.
6. Run ABL.

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Feb 18 2023 09:24:24)

### Printer model

MakerBot Replicator 2X

### Electronics

BTT SKR V1.4 Turbo

### Add-ons

ZMAX endstop plugged into ZMAX plug

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10774516/Marlin.zip)

thisiskeithb, 2023-02-18:
Enable `ENDSTOPS_ALWAYS_ON_DEFAULT` and see if homing behavior changes.

With `USE_PROBE_FOR_Z_HOMING` enabled, the Z endstops are ignored while homing & a probe is used instead.

Poikilos, 2023-02-20:
> Enable `ENDSTOPS_ALWAYS_ON_DEFAULT` and see if homing behavior changes.

That did not solve the problem.

> With `USE_PROBE_FOR_Z_HOMING` enabled, the Z endstops are ignored while homing & a probe is used instead.

Even if I turn off `USE_PROBE_FOR_Z_HOMING`, UBL doesn't work at all: It seems to reach hysteresis as usual, Z starts decreasing, then the sensor touches the bed, then Marlin says "T0 heating failed". It only happens when using BLTouch without `USE_PROBE_FOR_Z_HOMING` and only happens when the sensor touches the bed, even though the error is totally unrelated. With any version I previously tried of of bugfix-2.0.x or bugfix-2.1.x, I always had to enable `USE_PROBE_FOR_Z_HOMING` for UBL to complete.

`USE_PROBE_FOR_Z_HOMING` should only prevent the ZMIN from being used. The current behavior of also ignoring ZMAX is a bug since it causes gears to grind by ignoring a sensor unrelated to the probe. Only ZMIN is related to the probe.

Poikilos, 2023-02-21:
I also confirm that the custom Z_MAX endstop works, as it triggers at the end of a large print when build volume is set to taller than the hardware (Marlin sends an endstop triggered message to the BTT TFT35 touch mode screen).

ellensp, 2023-02-24:
The skr 1.4 does not have official max endstops

It has:
```
x-stop | E0DET
y-stop | E1DET
z-stop | PWRDET
```

Now in special cases marlin will juggle the pins about and use some of these other pins as max endstops.
Your configs do not trigger this. 
On Z this only happens if your using sensorless or dual z endstops

So with your config marlin has 
    #define Z_STOP_PIN                     P1_27  // Z-STOP

Since marlin only knows about z-stop that is assigned to z-min if homing to min or z-max if homing to max, even if it is not used because you using the probe to home.

Marlin still doesn't know where you have plugged this Z_MAX endstop into?

And neither do we

So the question is, where did you plug in the Z_MAX endstop?

Poikilos, 2023-02-24:
Ah, thanks for the response. It is plugged into PWRDET, the one next to ZMIN.

ellensp, 2023-02-25:
try this:

In pins_BTT_SKR_V1_4.h

replace

`#elif ENABLED(Z_MULTI_ENDSTOPS)
`

with

`#elif EITHER(Z_MULTI_ENDSTOPS, USE_ZMAX_PLUG)
`

Marlin will then define 
    #define Z_MIN_PIN                      P1_27  // Z-STOP
    #define Z_MAX_PIN                      P1_00  // PWRDET

instead of just
    #define Z_STOP_PIN                     P1_27  // Z-STOP

I don't consider this is a bug, its just a lack of realizing that this board does not actually have max endstops, so you have to force it to use PWRDET as Z_MAX_PIN

closed 2023-02-25 by ellensp

Poikilos, 2023-03-01:
I made the change to the pins and the problem persists. I don't know how it was even stopping at 149 before I made the change, but it stopped there or 140 so I don't even know if that is possible that is was a software endstop since the number differed. In any case, it was only triggering and stopping during a print (end gcode to lift to 150) and not triggering during UBL homing, with or without the change to pins.

firmware.bin got renamed to FIRMWARE.CUR but it didn't ask to reset firmware settings. Do I need to do that?

Poikilos, 2023-03-14:
@ellensp Can you comment on my response? I tried your solution.

Poikilos, 2023-04-08:
It is not working. Is there anywhere to look? It still happens on UBL but the physical endstop works on movement.

Poikilos, 2023-04-19:
@thisiskeithb Can you comment on my reply above? I realized today I should try this:
[image]
It still grinds on auto home even with MAX_SOFTWARE_ENDSTOP_Z commented. Please fix this (implement this allowed configuration so it works).

Poikilos, 2023-05-01:
I will provide any new information as I can. I have updated to the latest commit (2 days ago) on the bugfix-2.1.x branch. The new settings I'm using are:

Configuration.h
```C
// Min software endstops constrain movement within minimum coordinate bounds
#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
  #define MIN_SOFTWARE_ENDSTOP_X
  #define MIN_SOFTWARE_ENDSTOP_Y
  //#define MIN_SOFTWARE_ENDSTOP_Z
  #define MIN_SOFTWARE_ENDSTOP_I
  #define MIN_SOFTWARE_ENDSTOP_J
  #define MIN_SOFTWARE_ENDSTOP_K
  #define MIN_SOFTWARE_ENDSTOP_U
  #define MIN_SOFTWARE_ENDSTOP_V
  #define MIN_SOFTWARE_ENDSTOP_W
#endif

// Max software endstops constrain movement within maximum coordinate bounds
// #define MAX_SOFTWARE_ENDSTOPS
```

And I added custom code here:
```C
#if ENABLED(USE_PROBE_FOR_Z_HOMING) // Poikilos added this case
  #if DISABLED(MAX_SOFTWARE_ENDSTOP_Z) // Poikilos added this case
    #define CUSTOM_Z_MAX
  #endif
#endif

#ifdef Z_STALL_SENSITIVITY
  #define Z_STOP_PIN                  Z_DIAG_PIN
  #if Z_HOME_TO_MIN
    #define Z_MAX_PIN                      P1_00  // PWRDET
  #else
    #define Z_MIN_PIN                      P1_00  // PWRDET
  #endif
#elif ENABLED(Z_MULTI_ENDSTOPS)
  #ifndef Z_MIN_PIN
    #define Z_MIN_PIN                      P1_27  // Z-STOP
  #endif
  #ifndef Z_MAX_PIN
    #define Z_MAX_PIN                      P1_00  // PWRDET
  #endif
#elif ENABLED(CUSTOM_Z_MAX) // Poikilos changed to allow PWRDET on BTT SKR V1.4 to function as ZMAX (workaround invalid issue https://github.com/MarlinFirmware/Marlin/issues/25401#issuecomment-1445031951)
  #define Z_STOP_PIN                      P1_00  // PWRDET
  // NOTE: only setting Z_MAX_PIN (leaving MIN undefined) causes a compilation error.
#else
  #ifndef Z_STOP_PIN
    #define Z_STOP_PIN                     P1_27  // Z-STOP
  #endif
#endif
```
- Unfortunately, it will not let me define Z_MAX_PIN without also defining Z_MIN_PIN. If I only define Z_MAX_PIN, there is a compilation error. If I run it this way, the issue still occurs, and the output of M119 always says I have a z_min (it doesn't list a z_max), even though it is plugged into PWRDET and M119 correctly reports the state of the swich (but on z_min instead of z_max as desired).