MarlinFirmware/Marlin #25406: LCD_BACKLIGHT_TIMEOUT support for LCD's with Neopixel backlight

Daranbalt4, 2023-02-19:
### Describe the feature you want

Here's my modified Marlin 2.1.1 which supports LCD_BACKLIGHT_TIMEOUT with a neopixel lcd. You can define a timeouted color for neopixel, so it's your decision to turn completely off your screen, or just dim the backlight, or change the light to a different color. I also made a new sanity check to make the changes compatible. I can use this modified firmware with SKR 1.4 Turbo board and a Bigtreetech 128x64 Neopixel lcd.

Video: https://www.youtube.com/watch?v=X1zXucc0Q4E&t=1s

Modified files: 
configuration.h
marlinui.cpp
neopixel.h
neopixel.cpp
sanitycheck.h

thisiskeithb, 2023-02-19:
Since you already have this working in your fork, can you submit a PR?

Daranbalt4, 2023-02-21:
OK, I'll make a try. Sorry for being noob, I'm new to github, this is the first time I use it to share my code.

thisiskeithb, 2023-02-21:
[Contributing Code with Pull Requests](https://marlinfw.org/docs/development/getting_started_pull_requests.html) should point you in the right direction.

Daranbalt4, 2023-02-26:
@thisiskeithb tried to make a pull request, but my code failed the test. I don't know, how could I repair this. Error message: https://github.com/MarlinFirmware/Marlin/actions/runs/4277147699/jobs/7445764279
But the declaration is here: https://github.com/MarlinFirmware/Marlin/pull/25438/files#diff-2f2b9a3cc91451771e08d3fb7ab060e67379325676ffa61f8142173fc02d457c.

thisiskeithb, 2023-02-26:
> tried to make a pull request, but my code failed the test. I don't know, how could I repair this.

thinkyhead will fix it up when he reviews the PR.

Closing this feature request since you’ve opened a PR.

closed 2023-02-26 by thisiskeithb