MarlinFirmware/Marlin #25413: [FR] Legacy Z offset for manually adjusted print beds.

SlickNickeL, 2023-02-21:
### Describe the feature you want

Legacy Z offset for manually adjusted print beds.

EXPLANATION:
Marlin has successfully produced a Duplication mode M605 allowing 3 + print heads to operate simultaneously on an extend print bed. This allows for multipart manufacturing.
The physical constraints of such a printer, requires a manual bed leveling only, due to the extended plain of 3 print heads. All three or more print heads must coexist on the same Z-plain. The print bed is trammed to accommodate the full reach of the combined print heads, across this extended Z-plain.
No auto-level can be used. 
No Mesh based component can be used.
No bed probe for sampling can be used with accuracy for all heads, simultaneously.
Auto-tramming or EEPROM stored and adjusted 3 or 4 point tramming, has great risks, and opportunities to crash multiple print heads into the print bed. This is not acceptable.
What is left is a very simple, but proven concept (in Repetier) for establishing a level bed across a large Z-plain. 
ALCA-5 or MIC-6 cast tool plate must be used. 
Printer must be set with physical Z endstop 1-2 mm above print bed, at its Reference adjusting screw.
All print heads on the Z-plain are raised 10mm above bed and manually trammed using a 10mm precision block at the Reference, Pitch and Roll points according to the bed setup. This will roughly place the bed on a 3 point plain in reference to the print head Z-plain.
After homing G28, all print heads are parallel with the print bead and collectively 1-2mm above bed. Then Legacy Z offset would be used to complete and fine tune the first layer.

PROPOSED: Legacy Z offset.  
LCD display showing (Legacy Z offset: 0.000) 
Would apply only to T0 hotend since all movements and settings (other then flow and temp) are cloned from T0 to T1, T2, T3…
Must work with multiple hotends allowing Duplication Mode M603 without causing a sanity check. This should also work with a single head, but doesn’t need any addition computations with Mesh, autolevel or probes, since this is Legacy, and not intended to be used with them.
This value is cumulative and stored in EEPROM once entered, and selected.
A G28 command is immediately encountered once a value is entered, with the value then placing the Z-plain at its universal machine Z0.0. (Preferably off and away from bed )From here the printer can be powered down and Z-Plain re checked across the bed, noting any possible head contacts to bed.
A first layer template must be used for every manufactured product. 
This template is then run, printing a first layer across print bed. 
Babystepping can be used to adjust first layer thickness, noted and then checked with a micrometer. Subsequent changes are added or subtracted to the Legacy Z offset which always maintains a current running value from the last entry, and updated on entry, and a auto G28 with a store to EEPROM. 
This is a quick and easy way to dial in a Duplication manufacturing printer.

ADDITIONALLY:
A range limit could be incorporated of +or- 2mm past physical Z endstop, if an optical endstop is used.  Example: Original Build is ALCA-5 Tool plate with Glass plate set at physical Zendstop of 0.02 You need to remove the glass to use BlueTape or the aluminum bed itself, and a negative number would have to be entered into (Legacy Z offset -0.xxx) this would also violate the physical Z endstop, without turning them off to do so. A Optical Z endstop with a long blade base, will easily accommodate this and would be designed with this in mind.

CONCLUSION:
After any G28, the Legacy Z offset is calculated from EEPROM settings, and the machine Z 0.0 is established. If the program calls for first layer at .15 mm, then a .15 mm first layer should occur throughout the beds build plane. (Actual results will most likely be .12 to .17 across all of the first layers.) depends on lots of variables in bearings,rods and belts. The biggest variable will be in the users ability to understand and navigate the 3 point tramming process, and tune in the correct amount of pitch and roll, depending on the bed setup. Once a reasonable first layer thickness is established at Reference point bed adjuster (not to be adjusted or moved), and the Legacy Z offset is set, it’s a matter of pitch and roll to evenly apply these setting along the Z-plain for even thickness throughout.

My current tramming program is called a 3over3 point tramming. This involves placing one of the three print heads over its own designated tramming point. There exists an opportunity and a dilemma on some 3 point bed setups, where a 4th point could be used for fine tuning if the ROLL axis bisects the bed with almost equal amounts on both sides. A problem exists also, if one of the “WINGS” of the Role plane is theoretical, and not physical. Only a 3over3 can be used here. It would be better to reassign the Reference point to allow for the longest tip to tip roll plane. Pitch is always figured and corrected first, and then bed optimization is adjusted with the Roll adjuster.
Without the Legacy Z offset, it’s difficulty (not impossible) to set a multi-head Duplication printer up, for accurate multiple prints. I currently have one of our 3 head units using Marlin in Duplication mode. The problem is, any of the Z offsets won’t work with multi head printing, be it manual probe with Z offset. The only configuration that works is Babystepping to check as mentioned above and M206. There is no easy way to set M206 except for direct connection through direct pronterface and manually change. This process requires us to set up a Microsoft Surface Pro to make these changes. This would be easier accomplished through each individual machines LCD. 
Without some sort of adjustable precision Z offset, Marlin will struggle with additional future leveling features when they become usable, such as laser leveling or fixed optical length comparative, to current referenced focal point value.

Roxy-3D, 2023-02-27:
I am very interested in seeing the discussion of this idea.    Duplication modes for IDEX machines are inherently hobbled because the bed leveling can really only address one extruder.

ellensp, 2023-02-27:
If there was any non summitry in the bed heights, you could not use duplication, as the single z axis would simultaneously have to be at two different heights

SlickNickeL, 2023-02-27:
This Legacy Z could be used with IDEX, but wasn’t proposed for such a machine.2 print heads that move independently on the X-axis still share the same Z and Y and are fixed to those axis. This simply allows 2 print heads to “Mirror” the movements using opposing coordinates on the X axis, or for 2 exact prints, they would clone the X axis to both print heads. The Z axis could still use all of Marlins Z offset and probe functions by using 2, Z axis motor/steppers placed on each side of the X axis that the 2 print heads ride on. These are then averaged between the two heads as far as Z Offset, still using one head/probe to create a mesh or better yet, a tramming plane stored in EEPROM.  Current advanced commercial printers using IDEX can split the the bed parameters of the Z axis down the middle allowing for each Z-axis motor to “Average” the mesh between the two print heads. It’s not perfect, and still will not work well on a less than perfect build surface. A first layer height of .30 easily masked and fills in any flaws that may be encountered, and would require an extremely flat build surface such as ALCA-5 or MIC-6. IDEX rely on a well constructed machine using better than consumer grade off the shelf parts. If the mesh deviation is limited, and then split to two independent Motors allowing for a greater nullification of the mesh closer to the Z intersecting line, then good results can be accomplished using such a system for 2 print heads.In true Duplication Mode using 3 fixed print heads on the X axis, the Z-plane is calculated across all of the print head nozzle tips and creates a single plane. This plane is then trammed/parallel to the Bed surface. The bed surface in this case, can be cantilevered and dropped from the 3 print heads see https://www.facebook.com/groups/3Dtechtalk/permalink/6215121291866112/There is no ability to average a single point. The Z axis must be calculated as described in the FR. The duplication mode is a viable solution in producing quality manufactured identical parts, with a reduced physical footprint using multiple colors all at the same time. Ease of use and adjusting the first layer, are paramount in the success of Marlin firmware to manufacture product. The above FB picture is .15 first layer using 3 different colors of PLA from 3 different filament manufacturers. The variance across the 13”x13” MIC-6 ranges from .12 to .17 printed on blue tape. The reason for this FR is to be able to use the LCD to change the Z offset. Currently, it must be done with M206. This requires a considerable amount of time to connect to the printer USB to make this change. Sent from my iPad