MarlinFirmware/Marlin #25414: [BUG] Compilation BOARD_MKS_ROBIN_NANO with MKS_ROBIN_TFT24

yannick-alch, 2023-02-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I bought a kingroon KP3S Pro 3D printer with TFT24 v1.2 screen on it.
I want to compile marlin 2.1.2 (latest release at this time) with BOARD_MKS_ROBIN_NANO motherboard, MKS_ROBIN_TFT24 screen and TFT_LVGL_UI.
When I compile, I have this error : `Marlin/src/HAL/STM32/../../inc/SanityCheck.h:3097:6: error: #error "TFT_LVGL_UI requires TFT_RES_480x320."`
I understand that I have to use a screen with 480x320 resolution but mine is 320x240.

But with this repo [(marlin),](https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin%20Nano) it seems to use LVGL and it running well.

How can I make it compile ?

### Expected behavior

Compile marlin with MKS_ROBIN_TFT24 screen and TFT_LVGL_UI activated

### Actual behavior

Can't compile Marlin

### Steps to Reproduce

1) Download Marlin repo below
2) Extract it
3) Open it and compile as it
4) See error

### Version of Marlin Firmware

Marlin 2.1.2

### Printer model

Kingroon KP3S Pro

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10803933/Marlin.zip)

EvilGremlin, 2023-02-22:
It's not a bug... LVGL_UI is orphaned code now and needs a lof of work in general.
And repo you pointing at don't have LVGL_UI at all, it use upscaled u8glib UI (TFT_CLASSIC_UI)
So you can use CLASSIC_UI or COLOR_UI

ellensp, 2023-02-22:
"I understand that I have to use a screen with 480x320 resolution but mine is 320x240."
Then you did not understand
All the graphics for TFT_LVGL_UI is designed for a 480x320. period. end of discussion

The code you linked is for does not have any  LVGL code in it.
It is using the equivalent of   TFT_CLASSIC_UI - Emulated DOGM - 128x64 Upscaled

closed 2023-02-22 by ellensp

EvilGremlin, 2023-02-22:
@ellensp lol, i was faster!

ellensp, 2023-02-22:
@EvilGremlin Indeed.

yannick-alch, 2023-02-22:
Ok, so I don't understand.
So my question is, how can I reproduce the screen link in the first post with marlin ? (see picture below)

[image]
[image]

EvilGremlin, 2023-02-22:
You'll have to revamp a lot if LVGL_UI code and all graphics to make it sork on 320x240. The old precompiled stuff they have in that repo is not quite marlin and in actually in violation of marlin's licensing

jmz52, 2023-02-22:
> So my question is, how can I reproduce the screen link in the first post with marlin ?

You can't, unless you write it from scratch by yourself.
If I recall correctly the stock firmware is closed-source fork of Marlin 1.x and UI is using some third-partly proprietary library.
Source code was never released to public (because of that library) and MKS recreated their UI for Marlin2.0 using LVGL for 480x320 displays only.