MarlinFirmware/Marlin #25415: [FR] Endstop working in both ways

maxholgasson, 2023-02-22:
### Is your feature request related to a problem? Please describe.

I'm using the marlin fw with a BTT octopus pro in a diy machine. When moving the machine manually I need two endstops per axis in order not to destroy the machine. After a shutdown I don't want to move the positions which is needed for homing and software endstops..so hardware endstops are an absolute must due to safety.

### Are you looking for hardware support?

BTT octopus pro

### Describe the feature you want

The endstop input should work in both directions. So X-axis should stop at min and max position regardless of a software endstop/limit. The two switches are wired in series and therefore in software it's needed to store the current direction in order to detect which limit was triggered

EvilGremlin, 2023-02-22:
ENDSTOPS_ALWAYS_ON_DEFAULT

ellensp, 2023-02-22:
No config files, suspect you simply haven't enabled USE_XMAX_PLUG

maxholgasson, 2023-02-22:
Oh that would be great if I just made a mistake, I enabled USE_XMAX_PLUG and USE_XMIN_PLUG but it stops only in one direction.
Which config files do you need? Configuration and Configuration_adv.h?

thisiskeithb, 2023-02-22:
> Which config files do you need? Configuration and Configuration_adv.h?

Yes.

> The two switches are wired in series

Are they NC or NO?

Also, the Octopus Pro has plenty of dedicated endstop ports, so why not wire each endstop individually?

maxholgasson, 2023-02-22:
I started with a new firmware.
So far I enabled always n endstops and joystick usage

>Are they NC or NO?

Right now I'm just using a wire to test it but later it would be two NC in series.

The octopus has only 6 endstop ports if I'm not wrong and I have six axis. 
As a second board I would like to use my old 1.4 Turbo and this has also limited endstop ports.

but even if I would wire  each endstop individually, would that be possible at all? I found only some dual endstop settings in the firmware but this is used with a second motor and not for the same motor 

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/10807138/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/10807139/Configuration.txt)

thisiskeithb, 2023-02-22:
> The octopus has only 6 endstop ports if I'm not wrong and I have six axis.

It has 8 endstop pins

[image]

> but even if I would wire each endstop individually, would that be possible at all?

Of course. By uncommenting both `USE_*MIN/MAX_PLUG` options, you can enable both min & max endstop on an axis.

`ENDSTOPS_ALWAYS_ON_DEFAULT` will keep endstops active at all times & not just for homing.

maxholgasson, 2023-02-22:
Ah ok, I assumed that because it was like this in the manual
[image]

but still 8 are to less as I would need 12 inputs

maxholgasson, 2023-02-22:
If I understand ir right, the endstop actions are handled in the endstops.cpp

If a switch is triggered we're in this function
`void Endstops::update() {`

The direction is determined by this and the motor stopped by PROCESS_ENDSTOP_X(MIN);
```
  if (stepper.axis_is_moving(X_AXIS)) {
    if (stepper.motor_direction(X_AXIS_HEAD)) { // -direction
      #if HAS_X_MIN || (X_SPI_SENSORLESS && X_HOME_DIR < 0)
        PROCESS_ENDSTOP_X(MIN);
```
As soon as we change the direction, we're in the else part and are able to move the motor again..
..what's missing now is a second query if the switch is triggered again and with the know direction we know in the following that the second limit of the way is reach. So we can give way for the other direction again.

I just don't know how to do that is the code is so complex of Marlin.

ellensp, 2023-02-22:
You cannot have a single endstop triggered from both directions.

Consider this scenario

You power on.  The endstop is already triggered. 
Which way is safe to move?  
You cannot tell. As there is no way to tell if the axis is an minimum or maximum position.

closed 2023-02-23 by thisiskeithb

maxholgasson, 2023-02-23:
Please re-open the request as it's not yet solved

This is possible, that's exactly how it's working on my milling machine with another controller and firmware.
Maybe they store the last position in an EEPROM or just do nothing till you manually release it in either way.
Actually that never happend so far, so I guess it would be rare in real life.
True is, that I have only one input per Axis anf it's working

ellensp, 2023-02-23:
most 3d printer home X and Y at the end of each print.
Real cnc machines have encoder hardware. So they instantly know where the motors are on power up. They don't use endstops for homing, they use endstop to stop you breaking the machine with gcodes that move things out of bounds.

maxholgasson, 2023-02-23:
https://planet-cnc.com/configure-limit-switch-inputs-controller-planetcnc-tng-software/

Example 2 shows that it’s possible as a reference

ellensp, 2023-02-23:
Limit switches are not endstops.

maxholgasson, 2023-02-23:
Please let us not argue about words.
I have a real machine here assembles by myself and bought as a semi-professional kit.
It’s working.
The end stop on my BTT board with the Marlin FW is doing the same thing with my attached switch but only In one direction.
Actually I found in the endstops.cpp a way and it’s stopping now in both ways. Only thing is as long as one switch is triggered it won’t move.
I need now some variable to store the last direction. If that it’s known I know which direction is safe. That’s all I want and it’s how it’s working in my other machine.
Why do you want me to believe it’s not possible. Let’s find a solution. I would do it in another way but the board has to less inputs for Dedicated switches and others show that there is a way around.

maxholgasson, 2023-02-23:
Hello, yes as I'm wrote the other control probably stores it in an EEPROM. Which 100000 cycles and a smart way that's no problem. Even if you would turn day machine off for every single day 10 times..it would work around 27 years
I want to attach additional switches to input pins to be able two switch between various speeds e.g Or store, replay various moves. Also my smaller 1.4 Turbo board has less inputs. 12 Input pins only for end stops would also mean double of wires.
Maybe one way without an EEPROM would be to move let's say 1mm to the left. If still the endstop is trigger your're on the left end stop. Then we could move 2mm to the right, if it's free than allow to continue  to the right. That procedere is only needed after a restart for the first time when hitting an end stop

The code already has an option to uncomment the min or max plug but it just doesn't work for both. That's why my feature request to enhance this and make it possible

maxholgasson, 2023-03-11:
I attached now two switches to two inputs but can't stop the motor when hitting the max switch.
You told that it is posible with two inputs, can you tell me how?