MarlinFirmware/Marlin #25421: [BUG] M104 does not start heating the nozzle

DonBower, 2023-02-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have this Start code:
```
; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; preheat bed for PETG
M104 B{material_print_temperature_layer_0}; preheat extruder for PETG

G28 ; home
G29 ; abl
;*** Start Heating ***
M190 S{material_bed_temperature_layer_0} ; heat to setting
M109 S{material_print_temperature_layer_0} T0 ; heat to setting
```

However, when the print starts, the nozzle waits for the bed to reach temp before it starts to heat.

### Bug Timeline

Seems to have been happening for a while now.

### Expected behavior

I expect the Bed and Nozzle to start heating while the G28/G29  code executes the bed level process. 

### Actual behavior

It builds the Bed mesh and waits for the bed to reach temp before the nozzle begins to heat.

### Steps to Reproduce

1. Add a bltouch to ender 3 v2 with 4.2.7 board.
2. Add the code above to start code.
3. Start a print

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x-mfs (Nov 16 2022 10:29:29) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender 3 V2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Printer model

Creality Ender 3 V2

### Electronics

4.2.7 Board

### Add-ons

Dual Z, Micro-Swiss NG

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Firmware is from [Your Site](https://marlin.crc.id.au/firmware/Ender%203%20V2%20-%20v4.2.7%20Board%20-%20BLTouch/)
[image]

The-EG, 2023-02-24:
> M104 B{material_print_temperature_layer_0}; preheat extruder for PETG

You are using `B` instead of `S`. According to [the docs](https://marlinfw.org/docs/gcode/M104.html), `B` sets the **max** auto temperature, not the target temperature.

The behavior  you see is expected given you:
 - Set the target bed temp, but not wait
 - Set the max hotend temp, but not wait (which in this case does nothing)
 - Set the target bed temp and wait
 - Set the target hotend temp and wait

closed 2023-02-24 by thisiskeithb

DonBower, 2023-02-24:
Thank you for the answer.  Switched to S and all is right with the world.