MarlinFirmware/Marlin #25427: Part cooling fan (PWM) stops without change via M106 or M107

RolyBugs, 2023-02-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Geeetech PI3 ProB (GT2560 RevA+) with Marlin 2.1.2 installed and a part cooling fan added.
Also tested with release 2.1.x and 2.0.x with the same result.

The part fan either "Hunts" speeding up and slowing down and then gradually slows to a stop or simply stops while printing following the first M106 command. 
This manifests with small component prints which need cooling to stiffen each layer before printing the next.
The problem occurs when printed from either Repetier or SD card or manually setting the fan speed from the printer control panel.

Running the fan via Manual Control in Repetier-Host works as expected including keeping the fan running for an hour.
I have written a short script which starts the fan and simultaneously moves the X, Y and Z axis.
If I run this script from Repetier-Host the fan runs continuously whilst moving the axis as it should.

### Bug Timeline

Probably old as dependant on print features.

### Expected behavior

Fan to continue running until a new M106 or M106 command sent.

### Actual behavior

Fan speed is inconsistent without gcode directive.

### Steps to Reproduce

Run a print requiring constant cooling.

Attached are the configuration files being used. Lines commented with RHB are my amendments.
The gcode file will demonstrate the problem on my printer. It sets the fan to a fixed speed after the first layer.

[FanBug.zip](https://github.com/MarlinFirmware/Marlin/files/10830682/FanBug.zip)

### Version of Marlin Firmware

Marlin 2.1.2

### Printer model

Geeetech PI3 Pro B

### Electronics

Arduino 2560 Rev A+

### Add-ons

Print cooling fan (PWM).

### Bed Leveling

None

### Your Slicer

Slic3r

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-02-25:
First thing I see "M106 S127.5"   

S requires whole number 0-255

RolyBugs, 2023-02-25:
The fraction is produced by Slic3r (50% fan/PWM).
I have edited out the fraction from the code but the with the same result.
Just re-tested this (for my own peace of mind) and the fan stops for a few seconds after about 45 seconds and then stops completely after about 1 minute 45 seconds.
Also sending M106 S127.5 from the Repetier gcode command sets the fan running constantly.
The fraction is reflected in the log (snippet below).
Thanks
R

...
17:45:06.222 : N17 M106 S127.5*78
...

closed 2023-04-27 by RolyBugs