MarlinFirmware/Marlin #25428: [BUG] (bug summary)marlin does not show an error during G28-Auto_home when the sensor is faulty

radex324, 2023-02-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Why marlin does not show an error during G28-Auto_home when the sensor is faulty. The machine performs "In order to improve positional accuracy, the homing procedure can re-bump at a slower speed according to the [XYZ]_HOME_BUMP_MM and HOMING_BUMP_DIVISOR settings." But it doesn't show an error when endstop doesn't click a second time. If it is still pressed, it just continues zeroing by setting the damaged axis to 0. And it should show an error. It used to work that if the position of the first click did not match the position of the second click, it would show an error.

### Expected behavior

axis basing error should pop up

### Actual behavior

faulty sensor causes zero position 

### Steps to Reproduce

1. disconnect or activate endstop with tape/finger.
2. enable the G28 axis basing
3.The damaged axis does not move and thinks it is in position 0
4.Printer ready to work in spite of damaged axis (possible impact after going out of working range)

### Version of Marlin Firmware

2.1.2

### Printer model

ender 3

### Electronics

SKR mini E3 V 3.0, SKR 1.4

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

GMagician, 2023-02-25:
You have not posted your configuration files. Maybe VALIDATE_HOMING_ENDSTOPS may help?

radex324, 2023-02-25:
[Desktop.zip](https://github.com/MarlinFirmware/Marlin/files/10831208/Desktop.zip)
VALIDATE_HOMING_ENDSTOPS is enable.

radex324, 2023-02-25:
uncommented  #define DETECT_BROKEN_ENDSTOP and now it works, the printer checks the switches. Why is this option not enabled by default?

ellensp, 2023-02-25:
Every thing is an option.  You think it should be enabled by default, 50% of others say it shouldn't be enabled by default. 
Ultimately if it makes the code lager, it is disabled. 
This is why you are strongly encouraged to read the Config files and tailor it to your needs.

closed 2023-02-25 by ellensp