MarlinFirmware/Marlin #25430: [BUG] No icons on Ender 3V2

Firexstorm, 2023-02-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm currently setting up my Ender 3 V2 (stock screen with DWIN_SET (Custom) and 9.ICO installed) with BTT SKR MINI E3 V3.0 and used the Marlin release 2.1.2.

Issue:
Marlin Release 2.1.2\Marlin\src\lcd\e3v2\common\dwin_set.h line 28:
`//#define USE_STOCK_DWIN_SET // Use the Creality stock DWIN_SET instead of Marlin's unified DWIN_SET by The-EG & thinkyhead
`
If `USE_STOCK_DWIN_SET` isn't defined, the screen will have no icons at all and a problem with booting. 

`#define USE_STOCK_DWIN_SET` will fix this issue if you're using the stock display with the stock/costum firmware with the 9.ICO file for the icons.

### Bug Timeline

New bug

### Expected behavior

Icons on the main screen of the E3-V2 stock display

### Actual behavior

No icons on the main screen, only the text "Print", "Prepare", "Control",...

### Steps to Reproduce

1. use premade Configuration.h and Configuration_adv.h for Creality E3V2, BTT SKR MINI E3 V3.0 with CrealityUI
2. `#define NO_CONTROLLER_CUSTOM_WIRING_WARNING` in Configuration.h
3. compile with platformIO or Auto Build Marlin
4. Use costum wiring for the E3V2 screen as described in:
Marlin Release 2.1.2\Marlin\src\pins\stm32g0\pins_BTT_SKR_MINI_E3_V3.0.h line 140-186
5. load the firmware on the board and plug in the screen
-> No icons on the screen

### Version of Marlin Firmware

Marlin-2.1.2

### Printer model

Ender 3V2

### Electronics

BIGTREETECH SKR MINI E3 V3.0

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-02-27:
#define USE_STOCK_DWIN_SET // Use the Creality stock DWIN_SET 
//#define USE_STOCK_DWIN_SET // Use Marlin's unified DWIN_SET by The-EG "

You did not upload the unified DWIN_SET into the LCD, so you get no graphics

This is the expected behavior.

"Updating the Display
To update the graphics and icons on the display:

Format an SD card using the FAT32 filesystem with 4K cluster size.
Copy the DWIN_SET folder and private folder to the SD card and insert the card into the slot on the back of the display unit.
Power on the machine and wait for the screen to change from blue to orange.
Power off the machine.
Remove the SD card from the back of the display.
Power on to confirm a successful flash."

closed 2023-02-27 by ellensp

Firexstorm, 2023-02-27:
I know that I haven't uploaded the unified Files. "(stock screen with DWIN_SET (Custom) and 9.ICO installed)"
And as other users might update their firmware without thinking about the screen software, I thought that a comment on that topic might help. At least a notification on changing the DWIN_SET could be helpful.