MarlinFirmware/Marlin #25432: [BUG][Or Not} Incorrect RAM Reporting for LPC1769

Poporacer, 2023-02-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to compile firmware for an SKR E3 Turbo and it appears that it is reporting the RAM incorrectly? Or I am not reading the specs correctly?
It looks like this board should have 48-64K RAM (I couldn't find the exact spec) But when I run Marlin Auto Build in VSCode, it is reporting the RAM as 31.97KB which isn't even close to what I "think" it is supposed to be. I am unable to install Input Shaping because it states there is not enough RAM.
I am using the most current bugfix-2.1.x as of today.

Is this a bug, user error, or me not understanding the specs?
Thanks

### Bug Timeline

New

### Expected behavior

I expect the RAM size to reflect what the board actually has.

### Actual behavior

The code is reflecting a RAM much smaller than the board specs

### Steps to Reproduce

1. Modify the Ender 5 Pro Example Configuration files (Configuration.h and Configuration_adv.h) to enable most features and change the board type to BTT_SKR_E#_Turbo and enable Input Shaping X and Y.
2. Compile the firmware
3. See error 

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x (Feb 25 2023 20:42:47)

### Printer model

Creality Ender 5 Plus

### Electronics

BTT SKR E3 Turbo with TFT35 display

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[image]
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/10833047/Configuration_adv.zip)

ellensp, 2023-02-26:
SRAM is divided into banks

```
The LPC17xx contain a total of 64 kB on-chip static RAM memory. This includes the main
32 kB SRAM, accessible by the CPU and DMA controller on a higher-speed bus, and two
additional 16 kB each SRAM blocks situated on a separate slave port on the AHB
multilayer matrix.
This architecture allows CPU and DMA accesses to be spread over three separate RAMs
that can be accessed simultaneously
```

so programs  only have access to 32KB  

Only some things can use the other 2 16K blocks, not normal code. 

Could the HAL be modified to use these 16K blocks?  I don't know. 

I know at least one  16 block is normally used as a USB buffer.

ellensp, 2023-02-26:
So definitely not a bug. A architecture limitation

closed 2023-02-26 by ellensp

Poporacer, 2023-02-26:
I think there is still a Bug even with this new information. I couldn't re-open this issue and not sure if this will be seen. If not, I will file a new issue.

On my other printer, it looks like input shaping uses ~2K RAM (X and Y each use 2560 bytes but both combined use 2588 bytes)
 
On my LPC1769 board **without** input shaping enabled, it uses 11140 bytes out of 32736 RAM (34%) 
With the info from my other printer, **with** input shaping enabled it should only be using ~14K RAM, well under the 32K....

But if I enable just X shaping, I get a RAM overflowed and it shows that it is overflowed by ~18K???
Without Input shaping enabled there is ~21K RAM free so if it is overflowing by 18K does that mean that input shaping on the LPC1769 takes up 39K RAM?
Or is there something I am not understanding?

[image]

[image]
Thanks

cbagwell, 2023-02-26:
The shaping buffers can easily reach that range depending on your config files.  Their size is based off of SHAPING_MIN_FREQ and SHAPING_MAX_STEPRATE.  For SHAPING_MIN_FREQ, the buffer grows with smaller values instead of higher.

If SHAPING_MIN_FREQ is not set then it will be based on your SHAPING_FREQ_X.  If SHAPING_MAX_STEPRATE is not set then it will be based on DEFAULT_MAX_FEEDRATE.

Are you sure all those related values are the same as your other config with smaller buffers?

Poporacer, 2023-02-26:
I am pretty sure, here is a DIFF

cbagwell, 2023-02-26:
What is your DEFAULT_MAX_FEEDRATE set to?

Poporacer, 2023-02-26:
I just found that and it looks like the example template should be corrected.... I have been using that feed I will submit a pull request...
I think this might be a little high? LOL

#define DEFAULT_MAX_FEEDRATE          { 8000, 8000, 15, 15000 }

I changed it to something more reasonable and it works

ellensp, 2023-03-01:
Been talking to a lpc1769 expert.

You can tell Marlin to use the unused 16k normally used for Ethernet buffer

replace
```
  shaping_time_t      ShapingQueue::times[shaping_echoes];
```
with
```
  shaping_time_t      ShapingQueue::times[shaping_echoes] __attribute__((section("AHBSRAM1"),aligned));
```

Your original Config still over flows this 16K reign

You now gt the error
```
ld: .pio/build/LPC1769/firmware.elf section `.AHBSRAM1' will not fit in region `ETH_RAM'
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: region `ETH_RAM' overflowed by 63636 bytes
```
But if you lower DEFAULT_MAX_FEEDRATE you can get it to fit on the 16k.

Eg
```
#define DEFAULT_MAX_FEEDRATE          { 2040, 2040, 15, 15000 }
```

Poporacer, 2023-03-01:
Thanks for the great info!