MarlinFirmware/Marlin #25433: [BUG] Ender3v2 infinite bootloop with inserted SD card

dmitrygribenchuk, 2023-02-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello. Have a crash with an inserted card when booting after compiling the last version on bugfix-2.1.x: b1f45b384b [cron] Bump distribution date (2023-02-24). Temperature sensors (bed and nozzle sensor shows 0)

### Steps to Reproduce

1) Insert SD card
2) Power ON
3) bug here
4) Remove card
5) Boot without card
6) Insert card

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender3 v2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-02-26:
> Don't forget to include
>  A ZIP file containing your Configuration.h and Configuration_adv.h.

You forgot to do this.

dmitrygribenchuk, 2023-02-26:
I'm not forgot, I can't((

When I try to drop it here - it save only local path like. Try to find out from another browser/pc

file:///tmp/marlin/Configuration.h.gz
file:///tmp/marlin/Configuration_adv.h.gz

dmitrygribenchuk, 2023-02-26:
https://user-images.githubusercontent.com/15670247/221399707-e97fb394-5884-4e40-911a-ad0a26635d79.mp4

thisiskeithb, 2023-02-26:
> When I try to drop it here - it save only local path like. Try to find out from another browser/pc

You can use a site like [Pastebin](https://pastebin.com/) and then paste links back here.

dmitrygribenchuk, 2023-02-26:
[Configuration.h.gz](https://github.com/MarlinFirmware/Marlin/files/10833252/Configuration.h.gz)
[Configuration_adv.h.gz](https://github.com/MarlinFirmware/Marlin/files/10833253/Configuration_adv.h.gz)

https://user-images.githubusercontent.com/15670247/221399813-e728e3a1-d557-4653-a57a-dcd49e7e0e9c.mp4

ellensp, 2023-02-26:
BOARD_CREALITY_V4 has lots of build environment options, what of the following did you use?

env:STM32F103RE_creality 
env:STM32F103RE_creality_xfer 
env:STM32F103RC_creality 
env:STM32F103RC_creality_xfer 
env:STM32F103RE_creality_maple

Anc what MPU chip is on your controller board?

dmitrygribenchuk, 2023-02-26:
Yep, my fault. The env is STM32F103RE_creality

dmitrygribenchuk, 2023-03-29:
Hello. I compiled firmware just now and the problem was fixed.

closed 2023-03-29 by thisiskeithb