MarlinFirmware/Marlin #25435: [BUG] Incorrect center of bed.

dmidge8, 2023-02-26:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

In both version 2.1.x and release 2.1.2, the bed center doesn't work as expected. 

I have the following configuration.h:
```
#define X_BED_SIZE 200
#define Y_BED_SIZE 200
#define CUSTOM_X_OFFSET 1
#define CUSTOM_Y_OFFSET 2
// Travel limits (linear=mm, rotational=°) after homing, corresponding to endstop positions.
#define X_MIN_POS (-X_BED_SIZE / 2 + CUSTOM_X_OFFSET)
#define Y_MIN_POS (-Y_BED_SIZE / 2 + CUSTOM_Y_OFFSET)
#define Z_MIN_POS 0
#define X_MAX_POS (X_BED_SIZE / 2 + CUSTOM_X_OFFSET)
#define Y_MAX_POS (Y_BED_SIZE / 2 + CUSTOM_Y_OFFSET)
#define Z_MAX_POS 200

#define BED_CENTER_AT_0_0
#define Z_SAFE_HOMING
#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT X_CENTER  // X point for Z homing
  #define Z_SAFE_HOMING_Y_POINT Y_CENTER  // Y point for Z homing
#endif
```

But the home start a bit outside of the bed. The home seems to be on the bottom left corner and can't have it centered. The Y limit grinds a bit.

### Expected behavior

I expected the coordinate system to be centered at the middle of the bed, and also to have the home position at the center. None of it seems to be the case.

### Actual behavior

It grinds on the bottom left of the bed.

### Version of Marlin Firmware

2.1.2

### Printer model

Prusa i3 mk2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10833728/Configuration.zip)

ellensp, 2023-02-26:
"and also to have the home position at the center." 
you dont have a delta, how do you expect it to home to the center of the bed, when the limit switches are at the ends of the xais?
Homiing only sets to a known position. It does not move beyond that 
unless you enable /#define HOMING_BACKOFF_POST_MM { X_BED_SIZE/2, Y_BED_SIZE/2, 0 }  // (linear=mm, rotational=°) Backoff from endstops after homing

dmidge8, 2023-02-27:
I am sorry, I am confused. I may misunderstand the macros definitions. But `HOMING_BACKOFF_POST_MM` seems to be only in Configuration_adv.h.

Besides, what would mean the `// Travel limits (linear=mm, rotational=°) after homing, corresponding to endstop positions.` comment in front of `#define X_MIN_POS` if not that it defines the offset of the (0,0) from X endstop?
I am still not sure that I understand what `#define BED_CENTER_AT_0_0` is about though, but I guess that it is to find the middle position from the size of the bed.

I really struggle to understand the macros, so I really appreciate your time to explain what I do wrong. There is quite a lot of the configuration that I do by trial and error to be honest, but this one, I couldn't figure it out...

dmidge8, 2023-02-28:
In case it is an useful information, I want to add that this configuration was working fine on a previous version of the branch 2.1.x. And there is a sanity check for the homing position: `static_assert(WITHIN(Z_SAFE_HOMING_X_POINT, X_MIN_POS, X_MAX_POS), "Z_SAFE_HOMING_X_POINT can't be reached by the nozzle.");` that checks that the position can be reach for the homing position. If I define the `X_MIN_POS` to `0 + X_OFFSET`, then the assert fails - which leads me to believe that the X_MIN_POS set to a negative value is correct.

dmidge8, 2023-02-28:
I wonder if the stem of the issue is not here, in Conditionals_post.h:
```
// Define center values for future use
#define _X_HALF_BED ((X_BED_SIZE) / 2)
#if HAS_Y_AXIS
  #define _Y_HALF_BED ((Y_BED_SIZE) / 2)
#endif
```
Shouldn't it be?:
```
// Define center values for future use
#define _X_HALF_BED (X_MIN_POS + (X_BED_SIZE) / 2)
#if HAS_Y_AXIS
  #define _Y_HALF_BED (Y_MIN_POS + (Y_BED_SIZE) / 2)
#endif
```

Or maybe it should that part that is the problem:
```
#define X_CENTER TERN(BED_CENTER_AT_0_0, 0, _X_HALF_BED)
#if HAS_Y_AXIS
  #define Y_CENTER TERN(BED_CENTER_AT_0_0, 0, _Y_HALF_BED)
  #define XY_CENTER { X_CENTER, Y_CENTER }
#endif
```
Corrected to?:
```
#define X_CENTER TERN(BED_CENTER_AT_0_0, 0, X_MIN_POS + _X_HALF_BED)
#if HAS_Y_AXIS
  #define Y_CENTER TERN(BED_CENTER_AT_0_0, 0, Y_MIN_POS + _Y_HALF_BED)
  #define XY_CENTER { X_CENTER, Y_CENTER }
#endif
```

ellensp, 2023-03-01:
bed size is bed size. not travel size. It shoudl not use min positionions.

X/Y_MIN_POS set up the bed so that X0,Y0 is physically at X0,Y0

dmidge8, 2023-03-01:
Thank you for your explanations! I still have troubles to understand what you mean, but I'll try to rephrase, so you can point where I misunderstood:
- X/Y_MIN_POS set the distance from the endstop Y/Y to the position (0; 0) of the bed.
- X_BED_SIZE contains the size of the bed in X.
So if I am correct, if I set the (0, 0) at the center of the bed, I should choose X_MIN_POS = -X_BED_SIZE / 2 + some offset to have the probe at the center, right?

I was reading the ternary condition right, but I guess I misunderstood the variables. So you mean that X_CENTER is not the X distance from the X endstop to the center of the bed, but rather the X distance from the front left corner of the bed to the X center of the bed? In that case, I fully understand why the ternary condition is correct. But then, I don't understand how to set the center at (0,0).

In the meantime, I reverted back to the 2.1.1 version, were my configuration was/is working correctly. I kept the other branch with the 2.1.2 version, in case I can upgrade later on if there is a fix. So far, even with the amount of hours on it, I couldn't figure out how to fix my configuration.

ellensp, 2023-03-01:
"X_MIN_POS = -X_BED_SIZE / 2 + some offset"

you set this to the distance from X0 to the X endstop
Endstops should be left of the bed so its X_MIN_POS = -X_BED_SIZE / 2 **-** some offset"

closed 2023-03-01 by ellensp