MarlinFirmware/Marlin #25439: [FR] Support the newer "TJC" screens on the Ender 3 V2 Gen4

xzero121, 2023-02-26:
### Is your feature request related to a problem? Please describe.

No icons, cut off text while using new V2 Display Board.

### Describe the feature you want

Support for 4th Gen V2 Display Board - "TJC Board"

### Additional context

Similar Issue for S1 TJC Board, This is non touch screen model, not sure if a separate request is justified, but wanted to make sure it was known.

ellensp, 2023-02-27:
Just found this unofficial page documenting the format of the tjc.tft file format https://github.com/UNUF/nxt-doc/blob/main/File%20Formats/TFT.md

It does seem to match what im seeing in the creality provided file

eoyilmaz, 2023-07-15:
I'm looking forward for this to be implemented

mister-vent, 2025-03-16:
The TJC displays are also used in newer models like the Ender 3 Max Neo in conjunction with the STM32 chips, so I'm definitely keeping an eye on this one. I've tried to search around for some solutions, but as far as I know, only one person (@mriscoc) has been able to successfully create an icon set for these newer display models, and it doesn't work with the normal Marlin UI.

mriscoc, 2025-03-16:
> The TJC displays are also used in newer models like the Ender 3 Max Neo in conjunction with the STM32 chips, so I'm definitely keeping an eye on this one. I've tried to search around for some [...]

Just to clarify, I didn't make an icon set for TJC, what I did was make the firmware compatible with this screen so that it could be used without modification.