MarlinFirmware/Marlin #25449: [BUG] G38.2 does not work on COREXY

aquaforum, 2023-02-28:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

G38.2 does not sense signal from probe 
I fixed it with this patch
```
diff -ur a/Marlin/src/module/endstops.cpp b/Marlin/src/module/endstops.cpp
--- a/Marlin/src/module/endstops.cpp	2022-12-19 22:42:28.000000000 +0200
+++ b/Marlin/src/module/endstops.cpp	2023-03-01 00:21:48.942400084 +0200
@@ -717,7 +717,7 @@
   #define UPDATE_ENDSTOP_BIT(AXIS, MINMAX) SET_BIT_TO(live_state, _ENDSTOP(AXIS, MINMAX), (READ_ENDSTOP(_ENDSTOP_PIN(AXIS, MINMAX)) != _ENDSTOP_INVERTING(AXIS, MINMAX)))
   #define COPY_LIVE_STATE(SRC_BIT, DST_BIT) SET_BIT_TO(live_state, DST_BIT, TEST(live_state, SRC_BIT))
 
-  #if ENABLED(G38_PROBE_TARGET) && NONE(CORE_IS_XY, CORE_IS_XZ, MARKFORGED_XY, MARKFORGED_YX)
+  #if ENABLED(G38_PROBE_TARGET)
     #define HAS_G38_PROBE 1
     // For G38 moves check the probe's pin for ALL movement
     if (G38_move) UPDATE_ENDSTOP_BIT(Z, TERN(USES_Z_MIN_PROBE_PIN, MIN_PROBE, MIN));

```
Explain, who knows why there is a check on COREXY and others? 
[config.zip](https://github.com/MarlinFirmware/Marlin/files/10855319/config.zip)

### Version of Marlin Firmware

2.1.2

### Printer model

Custom build COREXY

### Electronics

MKS_GEN_L_V2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-04-19:
It looks like this was first introduced in https://github.com/MarlinFirmware/Marlin/commit/0934563b97328184c4cfe5abdb730b0ad11b531e, but I'm not sure why it was restricted to exclude CoreXY/YX.

thinkyhead, 2023-04-19:
> I'm not sure why…

Perhaps `stepper.axis_is_moving` was only updated later, but `G38` was missed in that update.

closed 2023-04-19 by thinkyhead

thisiskeithb, 2023-04-19:
This will be fixed in https://github.com/MarlinFirmware/Marlin/pull/25713