MarlinFirmware/Marlin #25451: Change M605 documentation to include S2 (E#)

SlickNickeL, 2023-03-01:
### Describe the feature you want

M605 on Marlin’s standard Mcode information site currently does not include the parameters for selecting multiple heads for duplication mode. This is done with the (E1, E2, E3…) The E for number of extruders, should follow the S2 parameter and be explained. Example (M605 S2 E2) allows for 3 print heads fixed on the same X axis to print simultaneously. 
There is currently no explanation of this process.

ellensp, 2023-03-02:
Marlin is split over several github repositories

- [Marlin](//github.com/MarlinFirmware/Marlin) : The firmware source code. Authorized for use under the GPLv3 License.
- [Configurations](//github.com/MarlinFirmware/Configurations) : Example configurations for a wide variety of machines.
- [MarlinDocumentation](//github.com/MarlinFirmware/MarlinDocumentation): Jekyll source code for the [marlinfw.org](https://marlinfw.org/) website.
- [AutoBuildMarlin](//github.com/MarlinFirmware/AutoBuildMarlin): A Visual Studio Code extension used to configure, build and install Marlin Firmware.

Issues with Marlin Documentation should be created in the MarlinDocumentation repository

ellensp, 2023-03-02:
A Pr has been created to update this
This is what it now looks like.

[image]