MarlinFirmware/Marlin #25459: [BUG] (bug summary)\macros.h:418:20: warning: comparison between signed and unsigned integer

rickjs, 2023-03-04:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

In file included from e:\marlin2\marlin\src\inc\MarlinConfigPre.h:37:0,
                 from e:\marlin2\marlin\src\inc\marlinconfig.h:28,
                 from Marlin\src\gcode\feature\trinamic\M122.cpp:23:
e:\marlin2\marlin\src\core\macros.h: In instantiation of 'constexpr decltype ((lhs + rhs)) _MAX(L, R) [with L = int; R = unsigned int; decltype ((lhs + rhs)) = unsigned int]':        
Marlin\src\gcode\feature\trinamic\M122.cpp:50:64:   required from here
e:\marlin2\marlin\src\core\macros.h:418:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         return lhs > rhs ? lhs : rhs;

### Bug Timeline

?

### Expected behavior

Compile without this error

### Steps to Reproduce

Compiling using Platformio

### Version of Marlin Firmware

2.0.x bugfix(3/2/23)

### Printer model

Prusa MK3S - Einsy

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Chatgpt suggests...

This warning suggests that there is a comparison being made between a signed integer (lhs) and an unsigned integer (rhs). Since signed and unsigned integers are stored differently in memory, this can cause unexpected behavior and lead to bugs in your code.

To resolve this warning, you can modify the _MAX function in the macros.h file to use the same data type for both the lhs and rhs parameters. One possible solution is to add a type cast to convert the unsigned integer to a signed integer before making the comparison: c++

template<typename L, typename R>
constexpr decltype(auto) _MAX(L lhs, R rhs) {
  return lhs > static_cast<L>(rhs) ? lhs : static_cast<L>(rhs);
}

By adding the static_cast<L>(rhs) expression, we are explicitly converting the rhs parameter to the same data type as lhs, which will resolve the warning and ensure that the function works as intended.

However, it's important to note that modifying the macros.h file can have unintended consequences and cause other errors or issues. Therefore, it's recommended to test your changes thoroughly and seek assistance from the Marlin firmware community or developers if you're unsure about how to proceed.
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/10887214/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/10887215/Configuration_adv.h.txt)

sjasonsmith, 2023-04-01:
I've confirmed the issue and posted a PR to fix it in bugfix-2.1.x.

thisiskeithb, 2023-04-01:
Closing since there’s now a PR to fix this:
- https://github.com/MarlinFirmware/Marlin/pull/25613

closed 2023-04-01 by thisiskeithb