MarlinFirmware/Marlin #25461: [BUG] Ender 5 plus with #define DGUS_LCD_UI RELOADED cannot use the screen move without homing

Neo2003, 2023-03-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I just cannot move any axe from the screen on Ender 5 Plus without doing a home before.
I am in front of the printer, then I know where I can move. The firmware should not prevent me from moving X, Y or Z.

I am just probably miss a configuration setting but could not find any related to M84 or Homing that would prevent the screen from allowing to move without homing first.

### Bug Timeline

With Marlin 2.1 and DGUS-reloaded on the screen

### Expected behavior

Allow moving the head at anytime

### Actual behavior

I need to home before being able to move the head from the screen, I can move the head from Octopi but my PC is far from the printer.

### Steps to Reproduce

Cancel a print for example or do a full G29 UBL, then you cannot move the head from the screen, it says "you need to home".

### Version of Marlin Firmware

2.1 bugfix 02010300

### Printer model

Creality Ender 5 Plus

### Electronics

Silent creality card

### Add-ons

MicroSwiss direct drive

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10888858/Marlin.zip)

Bob-the-Kuhn, 2023-03-04:
Are you sure you've supplied the correct configuration.h file?  It has both NO_MOTION_BEFORE_HOMING and HOME_AFTER_DEACTIVATE disabled.  When testing the latest bugfix-2.1.x on my SKR PRO it works just fine.

thisiskeithb, 2023-03-05:
> it says "you need to home"

Can you provide a photo? This text doesn't exist in Marlin or the *now archived* [DGUS Reloaded UI](https://github.com/Desuuuu/DGUS-reloaded).

As Bob mentioned above, you shouldn't have issues with moving things around if you have `NO_MOTION_BEFORE_HOMING` & `HOME_AFTER_DEACTIVATE ` disabled.

Since DGUS Reloaded is now archived and no longer being developed, I'd recommend switching to the `IA_CREALTIY` variant of `DGUS_LCD_UI`.

Neo2003, 2023-03-05:
Sorry I was writing from memory, here is the picture
[image]
And yes I am sure the .h files I uploaded are the ones I used to compile. And I have both disabled
//#define NO_MOTION_BEFORE_HOMING // Inhibit movement until all axes have been homed. Also enable HOME_AFTER_DEACTIVATE for extra safety.
//#define HOME_AFTER_DEACTIVATE   // Require rehoming after steppers are deactivated. Also enable NO_MOTION_BEFORE_HOMING for extra safety.

I am not against moving to IA_CREALTIY if I can keep the DWIN_SET from DGUS-reloaded, the TinyMachine ones are awful.

ellensp, 2023-03-05:
All touch screen have limitations, they have the minimal subset of feature the original developers wanted and no more.

This lcd code does not look at either of these defines

It has this code

```
void DGUSRxHandler::Move(DGUS_VP &vp, void *data_ptr) {
  const int16_t data = BE16_P(data_ptr);
  const float position = dgus_display.FromFixedPoint<int16_t, float, 1>(data);
  ExtUI::axis_t axis;

  switch (vp.addr) {
    default: return;
    case DGUS_Addr::MOVE_SetX:
      axis = ExtUI::X;
      break;
    case DGUS_Addr::MOVE_SetY:
      axis = ExtUI::Y;
      break;
    case DGUS_Addr::MOVE_SetZ:
      axis = ExtUI::Z;
      break;
  }

  if (!ExtUI::isAxisPositionKnown(axis)) {
    dgus_screen_handler.SetStatusMessage(FPSTR(DGUS_MSG_HOMING_REQUIRED));
    return;
  }

  ExtUI::setAxisPosition_mm(position, axis);

  dgus_screen_handler.TriggerFullUpdate();
}
```

if you change this to the following, it should do what you want

```
void DGUSRxHandler::Move(DGUS_VP &vp, void *data_ptr) {
  const int16_t data = BE16_P(data_ptr);
  const float position = dgus_display.FromFixedPoint<int16_t, float, 1>(data);
  ExtUI::axis_t axis;

  switch (vp.addr) {
    default: return;
    case DGUS_Addr::MOVE_SetX:
      axis = ExtUI::X;
      break;
    case DGUS_Addr::MOVE_SetY:
      axis = ExtUI::Y;
      break;
    case DGUS_Addr::MOVE_SetZ:
      axis = ExtUI::Z;
      break;
  }

#if ENABLED(NO_MOTION_BEFORE_HOMING)
     if (!ExtUI::isAxisPositionKnown(axis)) {
       dgus_screen_handler.SetStatusMessage(FPSTR(DGUS_MSG_HOMING_REQUIRED));
       return;
     }
#endif

  ExtUI::setAxisPosition_mm(position, axis);

  dgus_screen_handler.TriggerFullUpdate();
}
```

Neo2003, 2023-03-05:
Many thanks, I will try this change

ellensp, 2023-03-05:
you cannot mix and match code and DWIN_SETS, they are a matched pair, so no you cannot use IA_CREALTIY with reloaded DWIN_SET

Neo2003, 2023-03-05:
Ok, then I will keep RELOADED then.
The message blocking the moves requires a lot more #if, the one provided is not enough. I will do the required changed and report back

Neo2003, 2023-03-05:
It works now :)
Here is the modified file
[DGUSRxHandler.zip](https://github.com/MarlinFirmware/Marlin/files/10891716/DGUSRxHandler.zip)
Many thanks for your support, Marlin 2.1 is fabulous.

thisiskeithb, 2023-03-07:
See https://github.com/MarlinFirmware/Marlin/commit/4b9bb85b1266e8febbcb1931c8427d410a05bf84

closed 2023-03-07 by thisiskeithb

thinkyhead, 2023-03-07:
I applied your changes and committed, @Neo2003. If that solves the issue then we can close this.