MarlinFirmware/Marlin #25464: [BUG] Icons and some text are missing on ender-3 v2 neo

Andlix, 2023-03-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I think it is the same problem as #23860.
When I update to version 4.2.2, all icons and some text are missing.

I build with the configuration for Ender-3 v2 neo, with tag release-2.1.2.

I thought that the resources for the display didn't fit, so I updated them from https://github.com/mriscoc/Ender3V2S1/tree/Ender3V2S1-Released/display%20assets. However, that made everything worse, both the delivered firmware and version 2.1.2 no longer worked. I tested the stock version and Giadej.

The display is a DACAI.

### Expected behavior

Icon and all text are displayed.

### Actual behavior

Icon and some text are not displayed.

### Version of Marlin Firmware

2.1.2

### Printer model

Creality Ender 3 v2 neo

### Electronics

Stock electronics

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10891441/Configuration.zip)

Andlix, 2023-03-05:
Now I take the files from "Configurations\config\examples\Creality\Ender-3 V2\LCD Files", now I see the icons, but the text in menu is missing.

Andlix, 2023-03-06:
Finally I found a proper configuration.
If I find the time, I'll create a pull request for MarlinFirmware/Configuration  in the next few days.

ThomasToka, 2023-03-08:
@Andlix could you please share your findings? Thx

Fido789, 2023-03-11:
@Andlix Yes, it would be very helpful to share a proper configuration.

chtaube, 2023-03-11:
I've ran into the same display issue with my Ender-3 v2 Neo and my "solution" was to build the firmware based on @mriscoc release at https://github.com/mriscoc/Ender3V2S1

The travel limits (endstops) on my E3v2Neo where slightly different, and I'm using a dual gear extruder with a different feedrate, that's why I've modified and compiled my own firmware based on his release. But apart from that no modifications where necessary.

The DACAI display can be reflashed with the stock firmware. It's in included in his release, too.

loop0, 2023-06-24:
I have the same issue with my ender 3 v2 neo, I built the latest marlin 2.1.2.1, I reflashed my display with the configurations for DACAI display from release 2.1.2.1. I get the icons but the text on the menus won't show. Would you care to share your solution @Andlix? Thanks

JaxGM, 2023-07-05:
Hi! I'm having the same issue. I was able to find this in the Configuration repo. https://github.com/MarlinFirmware/Marlin/issues/23621

I haven't been able to find the results of this in the most recent release but I hope someone else may find this helpful! 😊

JaxGM, 2023-07-05:
I'd thought I'd share the entirety of my research in the hopes this helps someone else. The issue seams to be from the screen being of a different variety. This issue is not localized to to this exact models; the [Ender 3 v2 has a similar issue](https://www.reddit.com/r/MarlinFirmware/comments/xjz9vz/glitched_marlin_ui_when_compiling_for_ender_3_v2/) of a seemingly similar root.

If you want to revert back to the original Creality firmware for the Ender 3 v2 NEO, here they are: (I, admittedly, have done this a few time myself)

- https://www.creality.com/pages/download-ender-3-v2-neo
- https://www.reddit.com/r/ender3v2/comments/ur1ew9/dacai_screen_stock_firmware/

JaxGM, 2023-07-06:
Inseam to have found a solution! Thanks to the amazing u/Soneliem on Redit, I have learned that the Pro firmware by mriscoc on GitHub. (https://www.reddit.com/r/ender3v2/comments/14sh1zp/marlin_dacai_screen_bug/?utm_source=share&utm_medium=ios_app&utm_name=ioscss&utm_content=1&utm_term=1)

Let me know if this works for anyone else here! Best of luck to all of you!

HyperDX, 2023-07-07:
Thanks! This firmware works flawlessly. All icons and texts are visible :).

closed 2023-09-17 by github-actions[bot]