MarlinFirmware/Marlin #25467: Possible bug in STM32 variants F4x7Vx HAL SOFTWARE_SPI

ricmm71, 2023-03-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Anet ET4 printer, STM32F407VGT6 cpu.
Maybe not a detectable bug, but potencially very annoying because some default pins are used without explicit configuration.
The default SPI object from library .platformio\packages\framework-arduinoststm32\libraries\SPI\src\SPI.h is usign default pins for SPI (file buildroot\share\PlatformIO\variants\MARLIN_F4x7Vx\variant.h), wich are not used for SPI in this board. But in file Marlin\src\HAL\STM32\HAL_SPI.cpp, method spiInit(), there is a call to SPI.begin(). Why to call to an object with no specific configuration? It could be a mess if some of the default SPI pins (from variant.h) where used in some STM32 board for other purposes and this function were called without other proper initialization.
In the case of ET4 printer there is no issue detected when commenting line 81 from Marlin\src\HAL\STM32\HAL_SPI.cpp :
<81> //    SPI.begin();

So it appears to be useless, at least, but I think it should be checked if it can be the cause for weird issues with SPI in others boards with MARLIN_F4x7Vx STM32 variants and enabled SOFTWARE_SPI.

### Version of Marlin Firmware

Marlin bugfix-2.1.x

### Printer model

Anet ET4

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

aldoguzman97, 2023-03-05:
I'm running SKR mini e3v3 stm32g0
I might be having similar issues. Auto home works movements works. As soon as a print is started. Only auto home works. Loading bar skips very quickly. Printer fails to stop print. 
Only happened after recent update to marlin.

jmz52, 2023-03-17:
> It could be a mess if some of the default SPI pins (from variant.h) where used in some STM32 board for other purposes and this function were called without other proper initialization.

It would mean that you are using wrong variant for your board.
These variants are not "one size fit all" and we would not need them if they were.
If your boards uses custom pinout, you need to create custom variant for it.
For example the `[env:STM32F407VE_black]` uses SPI2 on PB13-PB15 pins
https://github.com/MarlinFirmware/Marlin/blob/2.1.2/buildroot/share/PlatformIO/variants/MARLIN_F407VE/variant.h#L290

closed 2023-05-28 by github-actions[bot]