MarlinFirmware/Marlin #25471: Getting thermal runaway warnings from heating hot end

dmarnel, 2023-03-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When starting a print job, the hot end heats up well beyond the target temperature setting before it begins to cool down again. In previous Marlin versions the temperature would get to within a couple a degrees below the target temperature and pause before slowly heating to the target temp setting.

### Bug Timeline

I've only noticed this recently but haven't printed in several months.

### Expected behavior

I expect the temperature to rise to with a couple of degrees of the target setting, pause for the thermal check, and then resume heating until the target temperature is reached.

### Actual behavior

The hot end heats up anywhere from 3 to 6 degrees past the target temperature setting before it begins to cool down again.

### Steps to Reproduce

1. Start a print job.
2. Observe hot end heating past the target temperature setting.

### Version of Marlin Firmware

bugfix-2.1.x (Mar 5 2023 14:44:11)

### Printer model

Ender 3 Pro

### Electronics

BTT TFT35 LCD 

### Add-ons

Ender Direct Drive Extruder

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

dmarnel, 2023-03-06:
Here are the config files.
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/10900542/Archive.zip)

StevilKnevil, 2023-03-06:
Have you gone through PID tuning? https://reprap.org/wiki/PID_Tuning

dmarnel, 2023-03-07:
Did it this morning and I'm not seeing much of a change, if any. I have a job with a target temp. of 210, and as it heated the nozzle, the temp climbed all the way up to 204 then paused and went down to 196. In that time I received a thermal runaway warning through Octopod on my iPhone. It did climb back up and settled at 210 eventually.

togeis, 2023-03-12:
Same issue here, but mostly when doing a M109 R40 at my END gcode. Short before 40° it will raise the temp again with full power "@127" and then cool down again till 40°. Octoprint detects the short raise and sends me a "thermal runaway". Is can see this short raise also in the temp graph. I currently have no screenshot. I will reproduce the issue and will document this.

togeis, 2023-04-14:
Retested with bugfix release from 04-04-2023:

[image]

Still the same issue

ellensp, 2023-04-14:
@togeis your issue is not a thermal runaway issue. you should open your own issues.

klavatron, 2023-05-31:
**Marlin 2.1.2.1 with Arduino Mega2560 + Ramps**

When I set the extruder temperature to 245°C, the temperature only reaches 235°C before the extruder turns off for 1-2 seconds (I can see this from the LED indicator). Afterwards, it turns back on again. The firmware detects this temperature drop as a malfunction and shuts down the printer. However, if I can detect this drop before the shutdown occurs, I can send a command to heat the extruder again, preventing the printer from shutting down.

In contrast, Marlin **2.0.7.2** does not have this bug and operates correctly.

togeis, 2023-05-31:
Simular issue during cooldown phase. Heats up again during cooldown. Maybe related to the same bug.

https://github.com/MarlinFirmware/Marlin/issues/25676

closed 2024-09-23 by github-actions[bot]