MarlinFirmware/Marlin #25473: [FR] Dynamically slow down the print if heater is unable to maintain target temperature

StevilKnevil, 2023-03-06:
### Is your feature request related to a problem? Please describe.

I have tuned my printer to be able to print at 200mm/s. However when performing long extrudes (e.g. >3minutes vase mode) I eventually get thermal runaway because the cold filament cooling the hotend faster than the heater can heater. Note that this is after PID tuning, it's merely the case that the printer is printing too fast for the heater. 

One solution would be to slow the configured print speeds (e.g. in the slicer), but that is undesirable as this is only an issue for extended periods of extrusion, whereas a mix of print speeds and travel moves etc give the nozzle enough time to regain the correct temperature.

### Describe the feature you want

To avoid failed prints, detect when the hot end temperature is significantly (~3 degrees?) lower than the target temperature and slow the overall print speed down to give the nozzle a chance of coming back to temperature.

thisiskeithb, 2023-03-07:
You may have better luck with `MPCTEMP`.

StevilKnevil, 2023-03-07:
@thisiskeithb Thanks, I'd not spotted that before. Every time I try and run the `306 T` auto tune I get a temperature error as the temp goes above 215 degrees (it even goes above 230 if I increase the upper limit). Is there something I'm not doing right? I appreciate that this isn't the correct place to discuss it, but if you have any info beyond the basic documentation I'd appreciate it

StevilKnevil, 2023-03-10:
FYI The combination of `MPCTEMP` and `ADAPTIVE_FAN_SLOWING` achieved what this FR was trying to address. I'm happy to close this as "Will not implement"

closed 2023-03-10 by StevilKnevil