MarlinFirmware/Marlin #25475: [BUG] TwoTrees BLU-5 Bluer Plus Marlin Menu Problems

helakejr, 2023-03-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In Marlin 2.1.2 built for the TwoTrees BLU-5 Bluer Plus 3D Printer on the Marlin E/X/Y/Z Offset Menu selected from the main menu. After using the Z+ or Z- the back icon displayed at the Bottom/Right Corner of the screen gets jumbled. The jumbled Icon still works and returns to prior screen.

In Marlin Bugfix 2.1.x I found the that the Marlin E/X/Y/Z Offset Menu bug is still present.  I also found that all menus now display one too many lines causing the last line to over-write the ICONS in the last row. You are not able to select any icons (forward, backward or return to previous menu), the last menu item is always selected. Menus are now unusable.
[image]
[image]
[image]
[image]
[image]
[image]

### Bug Timeline

New, Marlin 2.1.2

### Expected behavior

Marlin 2.1.2: I expect using the Z+/Z- options not to scramble the Return to Prior screen ICON.

Marlin 2.1x: I expect using the Z+/Z- options not to scramble the Return to Prior screen ICON. And I expect all menus to not have an option on the last line of the display so that the three ICONS can be used (Forward, Backward, Return to Prior Menu).

### Actual behavior

Marlin 2.1.2: 
1. From main menu, select Z offset option (click on Z)
2. In the offset adjustment menu, select the Z+ and Z- options

### Steps to Reproduce

Marlin 2.1.2: 
1. From main menu, select Z offset option (click on Z)
2. In the offset adjustment menu, select the Z+ and Z- options
3. Verify the "Return to Prior Menu" ICON is NO longer displayed correctly

Marlin 2.1.x 2023/03/06:
1. Repeat Marlin 2.1.2 steps 1-3
2. Select any menu option from the main menu
3. Verify the screen, displays on too many lines overwriting Option ICONS on the last line of display.
4. Verify ICONs are inoperable. Pressing any ICON selects last line of menu display.
5. Verify only way to return to prior menu is to select the 1st line.

### Version of Marlin Firmware

Marlin Bugfix 2.1.x, Marlin 2.1.2

### Printer model

TwoTrees BLU-5 Bluer Plus

### Electronics

Stock

### Add-ons

None

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-bugfix-2.1.x.TT_BlueR-Plus.zip](https://github.com/MarlinFirmware/Marlin/files/10903101/Marlin-bugfix-2.1.x.TT_BlueR-Plus.zip)
[Marlin-2.1.2.TT_BlueR-Plus.zip](https://github.com/MarlinFirmware/Marlin/files/10903117/Marlin-2.1.2.TT_BlueR-Plus.zip)

bustamantemarlon, 2023-03-08:
I have the same issue on the same printer on Marlin 2.1.2. Not sure if it's related but my touch screen also seems really sensitive and would randomly click into menus without any user interaction.

thisiskeithb, 2023-03-08:
> Not sure if it's related but my touch screen also seems really sensitive and would randomly click into menus without any user interaction.

Enable `ENCODER_NOISE_FILTER`.

helakejr, 2023-03-08:
Thank you I'll check it out.

bustamantemarlon, 2023-03-09:
> Enable `ENCODER_NOISE_FILTER`.

Thank you for the suggestion. Unfortunately, this did not work for me. I even bumped up the sample to 15 with no change.

helakejr, 2023-03-14:
I discovered the Marlin 2.1.x bugfix code that broke the menus. In Marlin\src\inc\Conditionals_LCD.h
#if ANY(HAS_UI_320x240, HAS_UI_480x320, HAS_UI_480x272)
  #if ENABLED(TFT_COLOR_UI_PORTRAIT)
    #define LCD_HEIGHT TERN(TOUCH_SCREEN, 8, 9) // Fewer lines with touch buttons onscreen
  #else
    #define LCD_HEIGHT TERN(TOUCH_SCREEN, 6, 7) // Fewer lines with touch buttons onscreen
  #endif

was:
#if ANY(HAS_UI_320x240, HAS_UI_480x320, HAS_UI_480x272)
  #define LCD_HEIGHT TERN(TOUCH_SCREEN, 6, 7)   // Fewer lines with touch buttons onscreen

Unfortunately  the MKS_ROBIN_TFT43 touchscreen will not calibrate when set to "TFT_COLOR_UI_LANDSCAPE". 

Work around is to comment out new code.

helakejr, 2023-03-15:
The 2.1.x bugfix issue#25281 code change broke the Two Trees BLU-5 Bluer Plus mks tft43 menus.

helakejr, 2023-05-15:
Still an issue with the back screen Icon getting scrambled when Z+ or Z-
used.

helakejr, 2023-05-30:
Additional Detail:
- Z+/Z- causes problem
- "Soft Endstops" message causes problem
- Home "YZ" message causes problem
- Home clears problem by redrawing screen

helakejr, 2023-06-06:
Finally able to determine a fix for this issue. 
Here is a draft revised:  Marlin-2.1.2.TT_BlueR-Plus\Marlin\src\lcd\tft\ui_480x320.cpp file for my fix.
[ui_480x320.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/11668449/ui_480x320.cpp.txt)

thinkyhead, 2023-06-07:
It is always preferable if patches are sent using a Pull Request instead of attaching a file to a bug report. I apologize on behalf of GitHub that it's not easy to submit Pull Requests, but it really is the only guaranteed way that we get fixes into the codebase in a timely fashion. We attend to Pull Requests on a daily basis but it often takes us a long time to get around to reviewing issues. Nevertheless, I happen to be looking through Issues a bit today and just encountered this one by chance.

The patch may be straightforward, but since it is out of date with the current codebase it's a minor hassle for one of our volunteers to check out 2.1.2, compare the attached file, gather the changed lines, find the equivalent code in the current codebase, apply the solution to the current code, and then make a PR on your behalf. That said, I do appreciate that you took the initiative to figure out a solution.

You solution may not get patched into a 2.1.2 release, but —assuming it is well-crafted and without side-effects— it will be included in 2.1.3. That update won't be released for at least a couple of weeks. But once the patch is applied, you'll be able to test it by using the `bugfix-2.1.x` branch.

Thanks again. I'll let you know when the patch has been prepared and posted as a Pull Request for your additional review.

thinkyhead, 2023-06-07:
I've posted a PR for you to review over at #25942 so please give it a look and a thorough test to make sure it's all copacetic. It might also be good to have the author of the recent TFT code cleanup look at these changes to make sure they comport with his updates. We can continue the conversation over there, so I will go ahead and close this issue now.