MarlinFirmware/Marlin #25476: [BUG] Err: utf8 font not initialized

radkoko, 2023-03-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I print from SD card, all the screen text shows Err: utf8 font not initialized and if i try going to menu while printing the printer reboots.
It doesn't happen when I print via pronterface from pc, also it started after I updated FW after adding Z probe - now my FW takes almost all the space on motherboard (10 bytes left) - but I've been told it is OK.
When I start the print, printer goes through homing, warming up and bed levelling no problems and then when the actual print starts, it shows the filename and then the Error comes.
Also tried to format the SD card and use simple named gcode files.
[image]

### Bug Timeline

New bug, started after updating FW.

### Steps to Reproduce

1. Print from SD card
2. Go through heating, homing and bed levelling procedure.

### Version of Marlin Firmware

2.1.2

### Printer model

Creality Ender 3 pro

### Electronics

stock electronics

### Add-ons

microswitch Z probe

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/10908941/Marlin_config.zip)

DisasterofPuppets, 2023-04-12:
I get the same issue when printing from the SD card (usually around the 50% progress mark)
CR10-S Marlin 2.1.2
[(url)](url)
[image]
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/11206373/configs.zip)

thinkyhead, 2023-10-08:
This is likely a more general issue indicating possible memory corruption / buffer overflow. The symptom is caused by the variable `flag_fontgroup_was_inited` getting corrupted, most certainly a sign of memory corruption. Issues like this are hard to pin down and will manifest in a variety of ways.