MarlinFirmware/Marlin #25477: [BUG] - Setting "SERIAL_PORT_2 3" errors out due to conflict

c0dyhi11, 2023-03-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When setting `#define SERIAL_PORT_2 3` in `Configuration.h` the build fails due to "... conflict with another pin ..." (More info in output blow)

If I don't set `#define SERIAL_PORT_2 3` everything else compiles just fine, but I can't use the USB port on the board to connect to OctoPrint.

If I follow these same steps with branch `bugfix-2.0.x` these issues don't happen.

Error output is below in `Steps to Reproduce`below

### Bug Timeline

Validated this is still a problem with a fresh clone 10min ago, and I see a redit post from 10x days ago complaining. So I'm guessing this issue is around 2x weeks old.

### Expected behavior

That everything builds nicely.

### Actual behavior

The build fails with errors.

### Steps to Reproduce

1. Clone branch bugfix-2.1.x
2. Copy files from [here](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20Pro/CrealityV427) to the `Marlin` directory: 
3. Update `platformio.ini`:
```
default_envs = STM32F103RET6_creality
```
4. Update `Marlin/Configuration.h`:
```
#define SERIAL_PORT_2 3
```
5. Run the build:
```
platformio run --silent -e STM32F103RE_creality
```
6. Get Error:
```
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:50,
                 from Marlin/src/HAL/STM32/HAL.cpp:27:
Marlin/src/HAL/STM32/../../inc/../HAL/STM32/inc/SanityCheck.h:80:6: error: #error "Serial Port 3 TX IO pins conflict with another pin on the board."
   80 |     #error "Serial Port 3 TX IO pins conflict with another pin on the board."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:50,
                 from Marlin/src/HAL/STM32/HAL_SPI.cpp:27:
Marlin/src/HAL/STM32/../../inc/../HAL/STM32/inc/SanityCheck.h:80:6: error: #error "Serial Port 3 TX IO pins conflict with another pin on the board."
   80 |     #error "Serial Port 3 TX IO pins conflict with another pin on the board."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:50,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:27:
Marlin/src/HAL/STM32/../../inc/../HAL/STM32/inc/SanityCheck.h:80:6: error: #error "Serial Port 3 TX IO pins conflict with another pin on the board."
   80 |     #error "Serial Port 3 TX IO pins conflict with another pin on the board."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:50,
                 from Marlin/src/HAL/STM32/Servo.cpp:27:
Marlin/src/HAL/STM32/../../inc/../HAL/STM32/inc/SanityCheck.h:80:6: error: #error "Serial Port 3 TX IO pins conflict with another pin on the board."
   80 |     #error "Serial Port 3 TX IO pins conflict with another pin on the board."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:50,
                 from Marlin/src/HAL/STM32/eeprom_bl24cxx.cpp:32:
Marlin/src/HAL/STM32/../../inc/../HAL/STM32/inc/SanityCheck.h:80:6: error: #error "Serial Port 3 TX IO pins conflict with another pin on the board."
   80 |     #error "Serial Port 3 TX IO pins conflict with another pin on the board."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:50,
[... 88 lines cut ...]
v4.2.7 Board

### Add-ons

CR Touch (Creality Branded BL Touch) & Creality Filament Runout Sensor

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I didn't attach my `Configuration.h` and `Configuration_adv.h` as this issue is triggered with the default from [HERE](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20Pro/CrealityV427) with the one line modified in `Configuration.h` above.

ellensp, 2023-03-08:
The error is correct, Previous version of Marlin did no check for this conflict so many people got it wrong.

You have BOARD_CREALITY_V427

These are the two pins use for hardware UART 3

#define UART3_TX_PIN                        PB10  // default uses LCD connector
#define UART3_RX_PIN                        PB11  // default uses LCD connector

You have #define CR10_STOCKDISPLAY which uses the the LCD connector

This is in Conflict with the Encoder pins
  #define EXP3_03_PIN                       PB10
  #define BTN_EN1                    EXP3_03_PIN

If you wish to use Serial port 3, you should disable  the following
CR10_STOCKDISPLAY
SHOW_CUSTOM_BOOTSCREEN
CUSTOM_STATUS_SCREEN_IMAGE

ellensp, 2023-03-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-03-08 by ellensp

bob-anthony, 2023-03-08:
`SERIAL_PORT 1` should be sufficient to enable USB communication. Make sure your USB cable is data-capable and that you've defined HOST_ACTION_COMMANDS