MarlinFirmware/Marlin #25478: [BUG]/[FR]? Using a potomicrosensor reliably works for probing, but not in READ(Z_MIN_PROBE_PIN)

Sierra410, 2023-03-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

[I made an ABL probe](https://www.printables.com/model/417960-in-shroud-mounted-ender-3-bed-leveling-probe) which uses a photomicrosensor instead of a microswitch.  

While actual bed leveling works fine (the interrupts do fire), reading the state of the pin manually doesn't produce reliable results, which causes the PROBE_TRIGGERED_WHEN_STOWED_TEST logic to fail most of the time.  

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 3

### Electronics

Creality 4.2.2 motheboard

### Add-ons

https://www.printables.com/model/417960-in-shroud-mounted-ender-3-bed-leveling-probe

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I'm almost entirely sure that this is due to the fact that the sensor is an analog one and doesn't prevent the flow of current completely.  

While this could probably be fixed in hardware by adding a comparator, it seems that it could also be done in software.  

If I'm reading the documentation of STM32 right, the Z probe pin on Creality 4.2.2 is PB1, which appears to be capable of working in analog mode.  

Considering the fact that interrupts work fine already, it might make sense to add native support for probes that use analog sensors.  

For the time being I just made a quick and dirty workaround in probe.cpp.
```c++
  #if ENABLED(PROBE_TRIGGERED_WHEN_STOWED_NO_TEST)

    endstops.enable_z_probe(false);
    probe_specific_action(deploy);

  #elif ENABLED(PROBE_TRIGGERED_WHEN_STOWED_TEST)
```

Sierra410, 2023-03-09:
Oh. I really didn't think of EMI. I feel pretty stupid now.  

I assumed since it uses a pullup resistor it shouldn't be affected by currents induced by EMI much.  But, I think what's going on is that the phototransistor in my sensor just fails to pull the pin down completely, letting EMI mess with things.  

I just slapped a 100K resistor between the probe pin and ground. High enough to keep it above logic level, low enough to help the sensor pull the pin to ground.  

I think standard deviation of 0.001575 with 0.006 of range for a probe that shouldn't work reliably is pretty good. I'm happy with that.  

Thanks for help, @yet-another-average-joe!

closed 2023-03-09 by Sierra410

yet-another-average-joe, 2023-03-09:
Just had a look to my notes about the BLTouch (just unboxed), dated may 2019.

BLTouch std dev : from 0.0015 to 0.006 (ran 4 tests)

You're in the same ballpark. Just great results.

Numbers : 0.005506 , 0.001880 , 0.001027 , 0.001492

(this is repeatabilty of the sensor + repeatabilty of the machine ; the machine cannot be considered as perfect ; it is mind blowing)