MarlinFirmware/Marlin #25479: [BUG] Incorrect bed temperature measurements on MKS Robin Nano V3.1

commandos500, 2023-03-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Bed thermistor measures temperatures way lower then it should, triggering thermal runaway protection. 
I am using EPCOS 100kOhm thermistor setting and the temperature readings won't go higher then 70 C, even though the bed itself gets way hotter, and after a timeout it returns an error.
This behavior does not apply to the hotend

### Expected behavior

Bed temperature measures at least up to 110 C

### Actual behavior

Measurements stop at around 70-75 C and trigger thermal runaway protection

### Steps to Reproduce

1. Turn the the bed on with target temperature higher or equal to 80
2. Wait until the temperature stabilizes

### Version of Marlin Firmware

2.1.2

### Printer model

Geeetech A10M

### Electronics

MKS Robin Nano v3.1

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10918580/Marlin.zip)

Sonicboom247, 2023-03-11:
I am chasing the same issue, only thing i can find related is the MKS brand.  Mine is the MKS Monster 8 V2 with same issue on the bed. PID tuned to 70, start print with temp at 60 and temp fluctuates from 50 to 70 until thermal runaway and lost print. tested with both Cura and Prusa

inspectionsbybob, 2023-03-30:
MKS Robin_Pro, thought it was me when I went to see what my bed could do in advance of trying some fancy filaments.

closed 2023-08-09 by github-actions[bot]