MarlinFirmware/Marlin #25484: [FR] Update Variable Name BLTouch to a more generic name.

BrianBu01, 2023-03-08:
### Is your feature request related to a problem? Please describe.

As I was reading through the configuration I was searching for CR Touch but could only find BL Touch.  I am not an expert, but I believe the two devices operate on different principles.  I think CR Touches are optical sensors.

Anyways, I then had to search for configurations of machines that come stock with a CR Touch to confirm the BL Touch flag was the correct variable.

A short term fix could be fixed by adding a comment explaining both BL Touches and CR Touches use the same variable.  The variable name probably should be updated to TouchSensor, TouchDevice, ZAxisTouchSensor, BedLevelingTouchDevice or similiar.

BLTouch although is a partial acronym for the hardware type, it is recognized as a name for a specific brand and product.  This variable is representing more than just that one brand and product, and I would assume additional touch hardware will be released in the future.

### Describe the feature you want

Variable name update.

### Additional context

In naming variables there is a recommendation that when using acronyms, you should google the acronym and ensure your intended usage is the popular definition.  This helps avoid acronyms that are jargon or domain specific  to increase readability.

This acronym does come return the correct results, but the results are specific to just one brand/model.  Since this variable can be used with multiple brands/models the name should not be used.

My 2 cents.  

_"There are only two hard things in Computer Science: cache invalidation and naming things."_ -- Phil Karlton

thisiskeithb, 2023-03-09:
> A short term fix could be fixed by adding a comment explaining both BL Touches and CR Touches use the same variable.

Indeed. 3D Touches and other BlTouch clones would also fall into this category.

InsanityAutomation, 2023-03-09:
Considering the BLTouch is a patented device, and the clones are in clear violation of the patent regardless of antclabs lack of financial resources to stop them... I'm very much against renaming functions developed for first party devices to suit generic clones developed with missing features from reverse engineering the named product. I'm sure @Roxy-3D would agree that's quite a slap in the face to a company that's dealt directly with the development team here for years compared to companies actively ripping them off.

BrianBu01, 2023-03-09:
Any company that contributes to open source should realize their contribution will not be isolated to just their benefit.

This request to update the naming to represent the functionality, not a specific brand and product.

The variable is not being uses exclusively for one product.  Common coding standards would recommend naming types and members for the functionality, and to avoid specific brands and products. 

If there is a specific optimization the requires additional specificity that can only be found in one product, you probably should name that also after the specific optimization functionality.

This is not about ripping anyone off.  This is about writing clear and maintainable code.  Littering your code with artifacts from different companies that contributed does not help long term.

```
 /**
 * The BLTouch probe uses a Hall effect sensor and emulates a servo.
 */
//#define BLTOUCH

/**
 * Enable probes that emulates a servo to auto level the bed; This includes: BLTouch, CRTouch, (I personally don't know any others.)
 */
//#define ENABLE_BED_LEVELING_TOUCH_PROBE
```

thinkyhead, 2023-03-09:
So much documentation on the web, including from vendors of BLTouch clones, already explains that you should enable `BLTOUCH` that it would be problematic to change the name at this point. In essence, because they were the first out of the gate, the probe type itself is best known as "BLTouch-style." So, I don't think we should change the name to something more generic, regardless of the logical reasoning behind it.

closed 2023-03-09 by thinkyhead