MarlinFirmware/Marlin #25488: [BUG] Wrong pin mapping for MKS GEN L V2.1 (EEFB configuration)

amamchur, 2023-03-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Default pin mapping for MKS GEN L V2.1 (EEFB, Extruder Extruder Fan and Bed) doesn't match [PCB schematic](https://github.com/makerbase-mks/MKS-GEN_L/blob/master/hardware/MKS%20Gen_L%20V2.1_001/MKS%20GEN_L%20V2.1_001%20SCH.pdf) (see U_MOSFET
MOSFET.SchDoc)

### Bug Timeline

Unknown

### Expected behavior

#define HEATER_0_PIN                          10
#define HEATER_1_PIN                          7
#define HEATER_BED_PIN                        8
#define FAN_PIN                               9

### Actual behavior

#define HEATER_0_PIN                          10
#define HEATER_1_PIN                          9
#define HEATER_BED_PIN                        8
#define FAN_PIN                               4

### Version of Marlin Firmware

bugfix-2.0.x a6cc7a4f3584a6375f509584a2a5b24137b86561

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Config files here [cfg.zip](https://github.com/MarlinFirmware/Marlin/files/10936279/cfg.zip)

Main settings:
```cpp
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MKS_GEN_L_V21
#endif

#define EXTRUDERS 2

#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 1
//...
#define TEMP_SENSOR_BED 5

#define PIDTEMPBED

#define NUM_M106_FANS 1
```

borland1, 2023-03-11:
Have you tried the generic motherboard ( MKS_GEN_L )?  These MEGA boards are just Ramps 1.4 compatibles.  All the MKS Gen L pin mapping files (V1.0, V2.0, V2.1) include...

`#include "pins_RAMPS.h"`

I could be wrong, but I think the only differences is, V2.0 fixed PCB pin trace routing error for E1 Motor Driver plug-in module, and V2.1 added 3.3V/5V Motor Driver module options for 3.3V sensor-less homing.

How is schematic pin mapping different between V1.0, V2.0, and V2.1 ?

amamchur, 2023-03-11:
Hi, borland1.

I've already invastigated this issue and I cloud fix it my self.
But first of all I would like to clarify resolution with Marlin developers.

**Root of issue:** Marlin doesn't support EEFB (Extruder Extruder Fan and Bed) printer configuration for RAMPS compatible boards correclty.

MKS GEN L V2.1 is RAMPS compatible PCB with additional MOSFET (classic RAMPS have 3 MOSFETs).
This extra MOSFET provides ability to setup EEFB (Extruder Extruder Fan and Bed) printer configuration.

EEFB configuration requires MOSFET_D_PIN definiation. So if RAMPS board defines MOSFET_D_PIN it means they could support EEFB.

Full list of RAMPS compatible boards with EEFB support (Ctrl+F by MOSFET_D_PIN):
- 3Drag
- KFB 2.0
- ZUM Mega 3D
- Felix 2.0+
- MKS BASE
- MKS GEN L V2
- MKS GEN L V2.1
- MKS GEN L
- RAMPS S 1.2
- Sainsmart

**Technical details**

Here is macros (pins_RAMPS.h:235) which maps MOSFETs pin.
```cpp
#if FET_ORDER_EFB                                 // Hotend, Fan, Bed
  #ifndef HEATER_BED_PIN
    #define HEATER_BED_PIN          MOSFET_C_PIN
  #endif
#elif FET_ORDER_EEF                               // Hotend, Hotend, Fan
  #define HEATER_1_PIN              MOSFET_B_PIN
#elif FET_ORDER_EEB                               // Hotend, Hotend, Bed
  #define HEATER_1_PIN              MOSFET_B_PIN
  #ifndef HEATER_BED_PIN
    #define HEATER_BED_PIN          MOSFET_C_PIN
  #endif
#elif FET_ORDER_EFF                               // Hotend, Fan, Fan
  #define FAN1_PIN                  MOSFET_C_PIN
#elif DISABLED(FET_ORDER_SF)                      // Not Spindle, Fan (i.e., "EFBF" or "EFBE")
  #ifndef HEATER_BED_PIN
    #define HEATER_BED_PIN          MOSFET_C_PIN
  #endif
  #if EITHER(HAS_MULTI_HOTEND, HEATERS_PARALLEL)
    #define HEATER_1_PIN            MOSFET_D_PIN
  #else
    #define FAN1_PIN                MOSFET_D_PIN
  #endif
#endif

#ifndef FAN_PIN
  #if EITHER(FET_ORDER_EFB, FET_ORDER_EFF)        // Hotend, Fan, Bed or Hotend, Fan, Fan
    #define FAN_PIN                 MOSFET_B_PIN
  #elif EITHER(FET_ORDER_EEF, FET_ORDER_SF)       // Hotend, Hotend, Fan or Spindle, Fan
    #define FAN_PIN                 MOSFET_C_PIN
  #elif FET_ORDER_EEB                             // Hotend, Hotend, Bed
    #define FAN_PIN                            4  // IO pin. Buffer needed
  #else                                           // Non-specific are "EFB" (i.e., "EFBF" or "EFBE")
    #define FAN_PIN                 MOSFET_B_PIN
  #endif
#endif
```
Unfortunately, this macro was designed for 3 MOSFETS only, that's why Marlin doesn't support EEFB correctly.

**Suggested resolution.**

Add additional MOSFET order **_FET_ORDER_EEFB_** into pins.h. Currently this macro looks in the folowing way:
```cpp
#if NONE(FET_ORDER_EEF, FET_ORDER_EEB, FET_ORDER_EFF, FET_ORDER_EFB, FET_ORDER_SF)
  #if   MB(RAMPS_13_EFB, RAMPS_14_EFB, RAMPS_PLUS_EFB, RAMPS_14_RE_ARM_EFB, RAMPS_SMART_EFB, RAMPS_DUO_EFB, RAMPS4DUE_EFB)
    #define FET_ORDER_EFB 1
  #elif MB(RAMPS_13_EEB, RAMPS_14_EEB, RAMPS_PLUS_EEB, RAMPS_14_RE_ARM_EEB, RAMPS_SMART_EEB, RAMPS_DUO_EEB, RAMPS4DUE_EEB)
    #define FET_ORDER_EEB 1
  #elif MB(RAMPS_13_EFF, RAMPS_14_EFF, RAMPS_PLUS_EFF, RAMPS_14_RE_ARM_EFF, RAMPS_SMART_EFF, RAMPS_DUO_EFF, RAMPS4DUE_EFF)
    #define FET_ORDER_EFF 1
  #elif MB(RAMPS_13_EEF, RAMPS_14_EEF, RAMPS_PLUS_EEF, RAMPS_14_RE_ARM_EEF, RAMPS_SMART_EEF, RAMPS_DUO_EEF, RAMPS4DUE_EEF)
    #define FET_ORDER_EEF 1
  #elif MB(RAMPS_13_SF,  RAMPS_14_SF,  RAMPS_PLUS_SF,  RAMPS_14_RE_ARM_SF,  RAMPS_SMART_SF,  RAMPS_DUO_SF,  RAMPS4DUE_SF)
    #define FET_ORDER_SF 1
  #elif HAS_MULTI_HOTEND || (HAS_EXTRUDERS && HAS_CUTTER)
    #if TEMP_SENSOR_BED
      #define FET_ORDER_EEB 1
    #else
      #define FET_ORDER_EEF 1
    #endif
  #elif TEMP_SENSOR_BED
    #define FET_ORDER_EFB 1
  #else
    #define FET_ORDER_EFF 1
  #endif
```

Using **_FET_ORDER_EEFB_**  definition we can assign correct pins in pins_RAMPS.h:235.

borland1, 2023-03-12:
Glad you made progress finding a possible solution.

Any idea why these 8-bit MCU boards need a special RAMPS EFB configuration setup, which is unlike other newer 32-bit MCU boards?  Shouldn't the board's pin file be able to handle this without resorting to implementing the `RAMPS` pin file?

Is it the fact that the MCU Timer, that is suited for PWM control, is used for audio (Tone/Speaker)?

For this `MKS_GEN_L` board, I had previously reported a `Timer` glitch when `AUTO_FAN` is enabled, where any rotary dial user input would cause the extruder fan power to stop momentarily.   Report was closed without solution.  Other's recommended disabling `SPEAKER`.

amamchur, 2023-03-15:
Thx, borland1.

> Any idea why these 8-bit MCU boards need a special RAMPS EFB configuration setup, which is unlike other newer 32-bit MCU boards? Shouldn't the board's pin file be able to handle this without resorting to implementing the RAMPS pin file?

MKS_GEN_L has four MOSFET vs three MOSFET comparing to classic 8-bit RAMPS. First tree MOSFET could by easly mapped using the following definitions: FET_ORDER_EFB, FET_ORDER_EEB & FET_ORDER_EFF. Unfortunately, role of the last MOSFET isn't defined in pin.h file. I suggest to define additional MOSFET order FET_ORDER_EEFB (Extruder Extruder Fan and Bed), see my push request.

Using this definition we can easily map fourth MOSFET to appropriate pin in pins_RAMPS.h

```cpp
#elif FET_ORDER_EEFB
  #ifndef HEATER_0_PIN
    #define HEATER_0_PIN              MOSFET_A_PIN
  #endif
  #ifndef HEATER_1_PIN
    #define HEATER_1_PIN              MOSFET_D_PIN
  #endif
  #ifndef HEATER_BED_PIN
    #define HEATER_BED_PIN            MOSFET_C_PIN
  #endif
  #ifndef FAN_PIN
    #define FAN_PIN                   MOSFET_B_PIN
  #endif
#elif
```
> Is it the fact that the MCU Timer, that is suited for PWM control, is used for audio (Tone/Speaker)?
I think no.

> For this MKS_GEN_L board, I had previously reported a Timer glitch when AUTO_FAN is enabled, where any rotary dial user input would cause the extruder fan power to stop momentarily. Report was closed without solution. Other's recommended disabling SPEAKER.

Thx, for info. I'll keep it in mind.

borland1, 2023-03-15:
amamchur,

I guess I wasn't clear, as you seem to have missed my point.  The **MKS_GEN_L** board file is located in the Marlin **"RAMPS"** folder. And the "**pins_ramps.h**" file, that the `pins_MKS_GEN_L.h` board file calls  with `#include "pins_ramps.h"`, has almost 1000 lines of code.

There are other **Mega** MCU boards that have board pin files in Marlin's "**MEGA**" folder.  These boards do not use the same `pins_ramps.h` configuration, and in my view, are much simpler to configure.

I think it might be easier to generate a board pin file for the "**Mega**" folder.

amamchur, 2023-03-16:
borland1,

> I think it might be easier to generate a new board pin file for the "Mega" folder.

Agree. I've canceled my pull requests due to failed tests. My changes broke "Azteeg X3 Pro" configuration.
Please correct my if I'm wrong:
- Move pins_MKS_GEN_L_V2.h & pins_MKS_GEN_L_V21.h from "ramps" to "mega" folder
- Remove #include "pins_RAMPS.h" from pins_MKS_GEN_L_V2.h & pins_MKS_GEN_L_V21.h'
- Fix pin mapping
- From "Change log MKS Gen_L V2.1 change from MKS Gen_L V2.0"
    - TMC_E1_RX_PIN(SPI_CS) D21 to D12
    - Change servo1 pin D12 to D21

borland1, 2023-03-17:
I would search for a pin file in the MEGA folder to use as a template for creating a new pin file.  Take a look at file `pins_GT2560_REV_A.h`  you want to choose one that closely matches the hardware your using (limit switches, steppers, sensors, LCD display, speaker, encoder, etc.).  You can get the actual pin numbers for your board from the wiring diagram for the MKS_GEN_L provided by MKS.

You may need to edit the `inc/SanityChecks.h` file to eliminate the compile errors.

ellensp, 2023-03-17:
You guys are way over thinking this

All you need is

```DIFF
diff --git a/Marlin/src/pins/ramps/pins_MKS_GEN_L_V21.h b/Marlin/src/pins/ramps/pins_MKS_GEN_L_V21.h
index d508cb453f..1e4e957c43 100644
--- a/Marlin/src/pins/ramps/pins_MKS_GEN_L_V21.h
+++ b/Marlin/src/pins/ramps/pins_MKS_GEN_L_V21.h
@@ -35,7 +35,8 @@
 // Heaters / Fans
 //
 // Power outputs EFBF or EFBE
-#define MOSFET_D_PIN                           7
+#define MOSFET_B_PIN                           7
+#define FAN_PIN                                9
 
 //
 // CS Pins wired to avoid conflict with the LCD

```

closed 2023-03-20 by ellensp