MarlinFirmware/Marlin #25496: [FR] Allow MPCTEMP tuning to use the "Alternative Method"

StevilKnevil, 2023-03-11:
### Is your feature request related to a problem? Please describe.

My setup fails to autotune MPCTemp: https://github.com/MarlinFirmware/Marlin/issues/25481
However when I use the [Alternative Method](Manual Tuning
If M306 T doesn’t work –e.g., with positive temperature coefficient (PTC) hotends– MPC can be configured manually.), then it works OK

### Are you looking for hardware support?

No

### Describe the feature you want

`M306 T` should fail over to the alternative method of determining the parameters if the initial tuning fails (e.g. if the block or sensor responsiveness are negative).

It should also be possible to force the use of the Alternative Method (Or perhaps "differential method", as it's based on maximum rate of change?), maybe with `M306 TA`

tombrazier, 2023-03-13:
@StevilKnevil If you understand how MPC works, it would be great to see an alternative tuning method. I thought a lot about this some time back but did not have time to implement anything. An automated version of my manual tuning method may be a good start. (Can't remember off the top of my head if any of the long PR conversations included this but I can describe it necessary.)

StevilKnevil, 2023-03-14:
@tombrazier I understand the concept of MPC and am pretty happy with all of the tuning code there (oI did a bit of refactoring in the linked PR to help get my head around it!)

I also prototyped the alternative tuning mechanism and that worked fine for me, I just need to carve out a little time to tidy up that code and submit a PR.

PTC hotends should be a solvable problem, my only concern would be having the hardware to test on!

tombrazier, 2023-03-14:
I'm glad there is someone other than me who understands it and can work on it. I have a CHC hotend, which is PTC. I don't have a lot of time but I might be able to help test. Although I am interested that you seem to to be getting a negative sensor responsiveness if you don't have PTC. I had thought that this was entirely due to the higher part of the temp vs time curve being shifted to the left by PTC. But it's been quite some time since I looked at it.

StevilKnevil, 2023-03-14:
Yes, my hunch is that (for some reason) my block responsiveness is close to sensor responsiveness, breaking [assumption in the theory](https://marlinfw.org/docs/features/model_predictive_control.html#:~:text=In%20a%20real%20world%20system). (see: https://github.com/MarlinFirmware/Marlin/issues/25481).

While I have your attention could I also get your eyes on my [adjustments to the MPC documentation](https://github.com/MarlinFirmware/MarlinDocumentation/pull/487)? It caused me a bit of confusion when looking at manual tuning and comparing to the theory!

thisiskeithb, 2023-03-14:
Closing since you've opened a PR:
- https://github.com/MarlinFirmware/Marlin/pull/25517

closed 2023-03-14 by thisiskeithb