MarlinFirmware/Marlin #25501: [BUG] `M503 C` does not write configuration files to SD card.

CRCinAU, 2023-03-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When building a firmware with `CONFIGURATION_EMBEDDING` enabled, the configuration of that firmware is supposed to be written to the SD Card when `M503 C` is used.

Currently, no configuration files are written to the SD card.

```
Recv: Cap:CONFIG_EXPORT:1
...
Send: M503 C
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
...
Send: M22
Recv: echo:SD card released
```

No files are written to the SD card as expected.

### Expected behavior

As per documentation:
https://marlinfw.org/docs/gcode/M503.html

[C] 2.0.9.3 `CONFIGURATION_EMBEDDING` | Save the embedded configuration ZIP file to the SD Card or Flash Drive.

### Actual behavior

No files are written.

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-03-12:
is bailing on "if (!dirFile || isOpen()) return false" in Marlin/src/sd/SdBaseFile.cpp

!dirFill is failing and it just returns false

ellensp, 2023-03-12:
Add the failed check else shows that it fails to write, non silently.  
```CPP
        if (file.open("mc.zip", O_WRITE|O_CREAT) && file.write(pgm_read_ptr(mc_zip), size) != -1 && file.close())
          SERIAL_ECHO_MSG("Configuration saved as 'mc.zip'");
        else
          SERIAL_ECHO_MSG("Couldn't save mc.zip");
```

thinkyhead, 2023-04-03:
Maybe it would work to use `file.open(card.getroot(), "mc.zip", O_WRITE|O_CREAT)` instead…?

thisiskeithb, 2023-04-05:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/80839cf020b32971b4cb63f96a803c4fe19ce828?

ellensp, 2023-04-05:
Doesn't work.  Just created a new compile error

Using default config

enable
EEPROM_SETTINGS
SDSUPPORT
CONFIGURATION_EMBEDDING

add
FORCE_CONFIG_EMBED

compiling errors with

```
Marlin/src/gcode/eeprom/M500-M504.cpp: In static member function 'static void GcodeSuite::M503()':
Marlin/src/gcode/eeprom/M500-M504.cpp:72:64: error: no matching function for call to 'SdBaseFile::open(MediaFile, const char [7], int)'
         if (file.open(card.getroot(), "mc.zip", O_WRITE|O_CREAT) && file.write(pgm_read_ptr(mc_zip), size) != -1 && file.close())

```

CRCinAU, 2023-04-05:
Seems like a build error now:

```
Marlin/src/gcode/eeprom/M500-M504.cpp:72:64:
error: no matching function for call to 'SdBaseFile::open(MediaFile, const char [7], int)'
```

ellensp, 2023-04-05:
This will build, and creates the file. But the data within is not correct

```DIFF
diff --git a/Marlin/src/gcode/eeprom/M500-M504.cpp b/Marlin/src/gcode/eeprom/M500-M504.cpp
index 6da1d1cbd7..94fc36b0a9 100644
--- a/Marlin/src/gcode/eeprom/M500-M504.cpp
+++ b/Marlin/src/gcode/eeprom/M500-M504.cpp
@@ -66,10 +66,10 @@ void GcodeSuite::M502() {
 
     #if ENABLED(CONFIGURATION_EMBEDDING)
       if (parser.seen_test('C')) {
-        SdBaseFile file;
+        MediaFile file, root = card.getroot();
         const uint16_t size = sizeof(mc_zip);
         // Need to create the config size on the SD card
-        if (file.open(card.getroot(), "mc.zip", O_WRITE|O_CREAT) && file.write(pgm_read_ptr(mc_zip), size) != -1 && file.close())
+        if (file.open(&root, "mc.zip", O_WRITE|O_CREAT) && file.write(pgm_read_ptr(mc_zip), size) != -1 && file.close())
           SERIAL_ECHO_MSG("Configuration saved as 'mc.zip'");
       }
     #endif
```

thisiskeithb, 2023-04-15:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/25688

closed 2023-04-15 by thisiskeithb