MarlinFirmware/Marlin #25504: [BUG] Coordinates X, Y Crashes after run UBL bed leveling (ignores boundaries)

RKMARV, 2023-03-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi. 

I have installed to my flying bear ghost 5 printer bl-touch probe.
what we managed to do: the sensor is connected and installed
The probe works out and the systme shows values.

Problem: when I enter the command G29 P1 T, the probe starts to indicate points (about 82/100) and output the results to the terminal, BUT in random order, the whole head is together with the sensor for some reason
crosses beyond the boundaries of the table that along the XY axis. the result is :
1)Printer Head hitting to the random side forggeting about the boundries.
2)Topography represents values of points which are not the same to real position (again, it works in radom way).

I changed the mesh_inset parameters from 5 to 15 , did the mesh construction 30 times - the result is the same, in an arbitrary order when designating points, at a certain moment the printer head ignores the boundaries and measures with a probe outside the table.

I did tests when moving the head within the boundaries of the table with successive manual commands from terminal - all coordinates match.

Eeprom was reset every time a new firmware was loaded.

There is a feeling that some kind of glitch is happening (maybe the bltach is sensitive to other signals and you need to shield the wire - I don’t know).

Below are the configuration options:

#elif ENABLED(AUTO_BED_LEVELING_UBL)

  //===========================================================================
  //========================= Unified Bed Leveling ============================
  //===========================================================================

  //#define MESH_EDIT_GFX_OVERLAY   // Display a graphics overlay while editing the mesh

  #define MESH_INSET 5              // Set Mesh bounds as an inset region of the bed
  #define GRID_MAX_POINTS_X 10     // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle
  #define UBL_SAVE_ACTIVE_ON_M500   // Save the currently active mesh in the current slot on M500

  //#define UBL_Z_RAISE_WHEN_OFF_MESH 2.5 // When the nozzle is off the mesh, this value is used
                                          // as the Z-Height correction value.

// The size of the print bed
#define X_BED_SIZE 240
#define Y_BED_SIZE 195

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS -6
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE+X_MIN_POS 
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 200

#if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  #define MESH_MIN_X MESH_INSET
  #define MESH_MIN_Y MESH_INSET
  #define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
  #define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)
#endif

#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT (50)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT (5)    // Y point for Z homing when homing all axes (G28).
#endif

### Expected behavior

UBL bed leveling probes each point without glithing or crashed coodinates with boundaries.

### Version of Marlin Firmware

Version 2.0.7.2 

### Printer model

Flying Bear Ghost 5

### Electronics

Robin Nano 1.1 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/10951514/conf.zip)

DForck42, 2023-03-22:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11035561/Configuration_adv.zip)
having a similar issue with my ender 3 pro that's been modified to direct drive
skr mini e3 v2.0 board

XDA-Bam, 2023-03-26:
@RKMARV You have have set `#define PROBING_MARGIN 0`, which means that - unless your bed position is perfectly aligned to what the firmware thinks the bed position should be - your probe may end up on the edge of the bed or even outside of the bed. Technically, `#define MESH_INSET 5` should mean that there will be no probing within a margin of 5 mm around the bed.

But you also have `#define Y_MIN_POS -6` set and I'm pretty sure that there is currently a bug with negative values of `X/Y_MIN_POS` that I also encounter. This bug causes trouble when combined with low values `PROBING_MARGIN` in some cases and leads to probing outside of the bed area.

**Solution:** I'd recommend setting `#define PROBING_MARGIN 6` and `#define MESH_INSET 6`  (you may need to increase that to `7` or `8`) for now, until that potential bug gets fixed. I don't know about the random order of points - maybe that's normal, maybe there is another problem on your printer.

closed 2023-06-06 by github-actions[bot]