MarlinFirmware/Marlin #25505: [BUG] CMakeLists.txt is both ignored and shipped

rockstorm101, 2023-03-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

`.gitignore` file contains a line for ignoring files named "CMakeLists.txt". And therefore changes to files with such name are ignored. However file `buildroot/share/cmake/CMakeLists.txt` is shipped with the code.

### Expected behavior

No file named `CMakeLists.txt` is provided since it is ignored. Or if relevant it should be tracked.

### Actual behavior

File `buildroot/share/cmake/CMakeLists.txt` is shipped in the [latest release][1].

[1]: https://github.com/MarlinFirmware/Marlin/releases/tag/2.1.2

### Version of Marlin Firmware

2.1.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-03-13:
The file .gitignore lists

```
CMakeLists.txt
src/CMakeLists.txt

```
It does not list
`buildroot/share/cmake/CMakeLists.txt`

rockstorm101, 2023-03-13:
I guess I should've given the steps to reproduce the problem, my bad.

If I had three different `CMakeLists.txt` files in three different levels of subfolders, all of them will be ignored by a single line in `.gitignore`. Hope this helps visualize what I mean:

1. Three different `CMakeLists.txt` files, one on top level directory and the other 2 on subfolders
```shell
$ tree
.
├── CMakeLists.txt
├── dar
│   └── CMakeLists.txt
└── foo
    └── bar
        └── CMakeLists.txt

4 directories, 3 files
```
2. After a `git init` Git detects the three files:
```shell
$ git status
On branch master

No commits yet

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	CMakeLists.txt
	dar/
	foo/

nothing added to commit but untracked files present (use "git add" to track)
```

3. Add a single line to `.gitignore` saying "CMakeLists.txt":
```shell
$ echo "CMakeLists.txt" > .gitignore
```

4. **All** three "CMakeLists.txt" on all levels are now ignored by Git:
```shell
$ git status
On branch master

No commits yet

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	.gitignore

nothing added to commit but untracked files present (use "git add" to track)
```

If you only want to ignore the one on the top level folder then the right syntax on `.gitignore` is "/CMakeLists.txt".

thisiskeithb, 2023-04-28:
I just tested this on macOS 12.6.4/Monterey: `git version 2.37.1 (Apple Git-137.1)`:

If I create a `CMakeLists.txt` at the root of the project or under `src/`, no changes are detected.

Updating or deleting `CMakeLists.txt` under `buildroot/share/cmake/`, changes are detected.

Maybe some OS/git versions need a more explicit `.gitignore`?

closed 2023-07-09 by github-actions[bot]