MarlinFirmware/Marlin #25506: [BUG] (SKR 1.4 Turbo with BLTouch)

MarekStani, 2023-03-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

...
Mam drukarkę na bazie płyty BTT SKR 1.4 turbo. Chciałbym dodać do mojej drukarki sondę BLTouch.
Kiedy w Marlinie odkomentowuję #define BLTouch i próbuję zrobić kompilację pojawiają się błędy.
Próbowałem z wersją 2.1.2 oraz z aktualną bugfix.
Pierwszy z nich to: 
"compilation terminated.
*** [.pio\build\LPC1769\src\src\MarlinCore.cpp.o] Error 1
In file included from Marlin\src\feature\../module/../HAL/shared/servo.h:76,
                 from Marlin\src\feature\../module/servo.h:29,
                 from Marlin\src\feature\bltouch.cpp:38:
Marlin\src\feature\../module/../HAL/shared/../LPC1768/Servo.h:51:10: fatal error: Servo.h: No such file or directory"
...
[Raf.zip](https://github.com/MarlinFirmware/Marlin/files/10951664/Raf.zip)

### Bug Timeline

nowy. Pojawia się kiedy odkomentuję definicję BLTouch

### Version of Marlin Firmware

2.1.2 oraz najnowszy bugfix

### Printer model

HyperCube Evolution

### Electronics

SKR 1.4 turbo

### Add-ons

TMC2208, BLTouch

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

MarekStani, 2023-03-12:
Działa z wersją 2.0.9.5

ellensp, 2023-03-13:
Config files is bad

change to

#define DISABLE_Z false
#define Z_SAFE_HOMING

After that it complies without issue.

closed 2023-03-13 by ellensp

MarekStani, 2023-03-13:
Niestety to nie pomaga - nadal jest to samo.

ellensp, 2023-03-13:
delete directory  `C:\Users\{your user name}\.platformio` to and restart vscode to force a reinstall of platformio

MarekStani, 2023-03-13:
Dziękuję za pomoc - już działa.