MarlinFirmware/Marlin #25508: Sensorless homing needing Z_MIN_PROBE_PINw

Mikelodoncz, 2023-03-12:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I want to set up sensorless homing and probing, but marlin requires me to define Z_MIN_PROBE_PIN but why? When i cannot define probe pin beacuse I dont have a probe.

### Version of Marlin Firmware

2.1.2

### Printer model

Two Trees Sapphire Pro

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Marlin\src\module\endstops.cpp:402:28: error: missing binary operator before token "1"
  402 |   #if USES_Z_MIN_PROBE_PIN 1
      |                            ^
*** [.pio\build\mks_robin_nano_v1v2\src\src\module\endstops.cpp.o] Error 1
=========================================================== [FAILED] Took 287.52 seconds ===========================================================

Environment          Status    Duration
-------------------  --------  ------------
mks_robin_nano_v1v2  FAILED    00:04:47.523
====================================================== 1 failed, 0 succeeded in 00:04:47.523 ====================================================== 
PS C:\Users\User\Documents\Marlin FW\Marlin-2.1.x tt sp3> echo "done" >"C:\Users\User\AppData\Local\Temp\ipc"
PS C:\Users\User\Documents\Marlin FW\Marlin-2.1.x tt sp3> C

ellensp, 2023-03-13:
1) You shouldn't post bug reports when you haven't tried the latest code. 
2) You did not attach your configs as requested
3) sensorless still uses a IO pin.

So called sensorless uses the stepper driver stall detection feature. 
This uses the diag pin of the stepper driver that is going to be stalled.

ellensp, 2023-03-13:
4, sensorless probing is a really really bad idea.  It necessitates smacking your hotend into the bed.

This results in a) damaging your bed) b) mushrooming your hotend tip. (damaging extruder flow) 
This particularly should not be used for any machine that uses lead screws or threaded rods for Z

closed 2023-03-13 by thisiskeithb