MarlinFirmware/Marlin #25510: [BUG] Termocouple MAX31855 "BTT SKR3 EZ"  Error: TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN

qrstu, 2023-03-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With Board BTT SKR3 EZ I receive the following error: (Marlin 2.1.2)

arlin\src\HAL\STM32\../../inc/SanityCheck.h:2472:4: error: #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
 2472 |   #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\sdio.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:2472:4: error: #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
 2472 |   #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
      |    ^~~~~
Compiling .pio\build\STM32H743Vx_btt\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\STM32H743Vx_btt\src\src\HAL\shared\MinSerial.cpp.o
In file included from Marlin\src\HAL\STM32\tft\../../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\tft\gt911.cpp:26:
Marlin\src\HAL\STM32\tft\../../../inc/SanityCheck.h:2472:4: error: #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
 2472 |   #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\tft\../../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\tft\tft_fsmc.cpp:27:
Marlin\src\HAL\STM32\tft\../../../inc/SanityCheck.h:2472:4: error: #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
 2472 |   #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
 
 Was working for BTT SKR1.4 Turbo !!
 There file has been modified --> pins_BTT_SKR_common.h as follow. --> SPI-PIN

// CS, MISO, MOSI, and SCK for MAX Thermocouple SPI
//#if HAS_MAX_TC  //2021-07-03 Fix
  #define TEMP_0_CS_PIN   P0_26    // 2021-07-03 MAX31855 Thermocouple PIN Hotend
  #define TEMP_0_MISO_PIN P0_08     // 2021-07-03 MAX31855 Thermocouple PIN Hotend
  #define TEMP_0_SCK_PIN  P0_07    // 2021-07-03 MAX31855 Thermocouple PIN Hotend
//#endif      //2021-07-03 Fix

We had there long discussion about the fix.
--> https://github.com/MarlinFirmware/Marlin/issues/22289

TEMP_0_CS_PIN isn't available anymore in "pins_BTT_SKR_V3_0_common.h"

Please fix also for "BTT SKR3 EZ" Board or describe me a workaround that I can further use MAX31855 Thermocouple because BTT SKR 1.4 Turbo Board isn't available anymore. (End of Production) --> BTT SKR3 EZ is successor

Thanks

Regards
qrstu
 

### Expected behavior

Compile with out error.
Definition of TEMP_0_CS_PIN for BTT SKR3 EZ

### Actual behavior

Error compiling Marlin

### Version of Marlin Firmware

2..12

### Printer model

Creality Ender 3

### Electronics

BTT SKR3 EZ and BTT SKR1.4 Turbo

### Add-ons

TFT35

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

qrstu, 2023-03-13:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10953738/Marlin.zip)
Configs worked for BTT SKR1.4 Turbo --> Config modified for BTT SKR3 EZ --> Search for "2023-03-12" only Board changes and disabled Serialports will be investigated later.

ellensp, 2023-03-13:
You cannot just attach new hardware (ie a MAX31855 chip) and expect it to work

The  BTT SKR3 EZ does not come with a MAX31855 and there is no standard way to add one

So YOU have to add the required pin definitions

closed 2023-03-13 by ellensp

qrstu, 2023-03-13:
Which pins should be used because there is no SPI Port available.

EvilGremlin, 2023-03-13:
It must be hardware SPI, so you can use SD or display ports i guess

ellensp, 2023-03-13:
Or the SPI pins on the wifi esp port. 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).