MarlinFirmware/Marlin #25516: When compiled using CR10 S5 Configuration example extruder stepper doesn't work as expected [BUG] (bug summary)

akinilicali, 2023-03-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When you compile using the site provided configuration files and the bugfix-2.1.x version, the compilation finishes without errors. However, the extruder becomes unusably fast as a result.

### Bug Timeline

All recent bugfix versions.

### Expected behavior

I expect it to work in accordance with the e-step value. (93)

### Actual behavior

I1- Regardless of the value of the F statement in the g-code extrusion is always way too fast.
2- Extrusion continues for a very short while. 

### Steps to Reproduce

Issue command G1 E100 F100

### Version of Marlin Firmware

2.1.X Bugfix

### Printer model

Creality CR10 S5

### Electronics

Stock 2.2

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10962906/Configuration.zip)

ellensp, 2023-03-14:
Reset your eeprom to firmware defaults.
M502 then M500
Or use LCD Advanced Manu Initialize EEPROM

akinilicali, 2023-03-14:
That unfortunately doesn't solve the problem. I tried both methods you mentioned. When I use the Creality factory firmware which is a pre-compiled firmware as you know, the extruder works normally.

akinilicali, 2023-03-15:
These are the noteworthy differences between the configuration files of Bugfix from 15.03.2023 and those of the 2.1.2 release. 2.1.2 works OK, Bugfix is a no go.

Configuration.h
Bugfix 2.1.x
//#define DISABLE_X
//#define DISABLE_Y
//#define DISABLE_Z
//#define DISABLE_E  

Marlin 2.1.2
#define DISABLE_X false
#define DISABLE_Y false
#define DISABLE_Z false
#define DISABLE_E false

Configuration_adv.h
Bugfix 2.1.x
#define DISABLE_INACTIVE_X
#define DISABLE_INACTIVE_Y
#define DISABLE_INACTIVE_Z  // Disable if the nozzle could fall onto your printed part!

Marlin 2.1.2
#define DISABLE_INACTIVE_X true
#define DISABLE_INACTIVE_Y true
#define DISABLE_INACTIVE_Z true  // Set 'false' if the nozzle could fall onto your printed part!
#define DISABLE_INACTIVE_E true

Bugfix 2.1.x
#define MULTISTEPPING_LIMIT   16  //: [1, 2, 4, 8, 16, 32, 64, 128]

Marlin 2.1.2
No entry

Bugfix 2.1.x
#define ADAPTIVE_STEP_SMOOTHING

Marlin 2.2"
//#define ADAPTIVE_STEP_SMOOTHING

akinilicali, 2023-03-16:
I must set the record straight. When I use Marlin 2.1.2, the printer slows down to normal speeds and performs a print. However, when I issue this command sequence: M83 G1 F100 E100, extrusion is still cut short at 1.5 cm. Such under extrusion would normally result in a failed print. However, the printer can at least perform a print. In the case of Bugfix 2.1.x, the extrusion is both very fast and short and doesn't result in a working print. Now I am beginning to become suspicious of the mainboard.

akinilicali, 2023-04-03:
Thank you for the clarification.

closed 2023-06-14 by github-actions[bot]